autowarefoundation/autoware #466: Dfiierence between driving and passing in waypoint_loader

sgzzgit, 2016-10-05:
Please introduce the Dfiierence between driving and passing in waypoint_loader.

sgzzgit, 2016-10-05:
sorry to bother again.Please help me thanks! @h-ohta

h-ohta, 2016-10-05:
@sgzzgit 
waypoint_loader can publish two waypoints,driving waypoints and passing waypoints as LaneArray from app tab.
You can change two waypoints from lane_select app tab.

sgzzgit, 2016-10-05:
Thanks for your reply.
    I don't understand the difference  between driving waypoints and passing waypoints when navigating.
During my real vehicle test, we need to make the vehicle follow the waypoint array generated from 'waypoint_saver'. However there are two waypoints array in the same lane, as you can see in the picture below. 
when the vehicle run to this lane ,the 'path_select' module will select the nearest waypoint array so the steering wheel shakes severly. Could you tell me  how to make the vehicle follow only one waypoints array under this situation?
Thanks very much @h-ohta

h-ohta, 2016-10-06:
The difference between driving and passing is simply the difference of the name.
In other words, you can publish two waypoints from `waypoint_loader` and `lane_select` can select one of two.
By the way, `path_select` only relay the topic `/temporal_waypoints`.

If you use latest Autoware components, the vehicle drives on only one lane, and change lane from `lane_select` manually.
I'm sorry but I cannot understand your problem yet.

Please show me the detail information of your problems? video and so on...
Thank you.

sgzzgit, 2016-10-06:
Thanks for your reply.
I have recorded a video please check it and you will know the problem.

sgzzgit, 2016-10-06:
Here are screenshoots of the video.
[image]
[image]
[image]

sgzzgit, 2016-10-06:
velocity_set module will choose the nearest waypoint as next waypoint, however we want the vehicle run follow one of the specific lane, no switching.

closed 2016-10-07 by sgzzgit

h-ohta, 2016-10-07:
Thank you for your video.
I watched it, and I wonder you have two lane data in one file?
If so, you devide two lane data into individual files.
If you can, could you show me your file of the path?

sgzzgit, 2016-10-08:
Thanks for your reply. I have solved the problem already.
Many thanks!