autowarefoundation/autoware #471: Cant replicate quick start video

chiyuan-goh, 2016-10-10:
Followed exactly the steps in the English manual, but when I unpause the bag file after launching my_map.launch and my_localization.launch, I encountered the following

```
ndt_matching: /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:430: static void Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::run(Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, const Lhs&, const Rhs&, Packet&) [with Lhs = Eigen::Block<const Eigen::Matrix<double, 6, 6>, 6, -1, true>; Rhs = Eigen::Matrix<double, -1, 1, 0, 6, 1>; Packet = __vector(2) double; int LoadMode = 1; Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index = long int]: Assertion `lhs.cols()>0 && "you are using a non initialized matrix"' failed.
[ndt_matching-6] process has died [pid 2797, exit code -6, cmd /home/foo/catkin_ws/devel/lib/ndt_localizer/ndt_matching __name:=ndt_matching __log:=/home/foo/.ros/log/d03f04ce-8eb0-11e6-9c56-408d5cb83aa0/ndt_matching-6.log].
log file: /home/foo/.ros/log/d03f04ce-8eb0-11e6-9c56-408d5cb83aa0/ndt_matching-6*.log
```

It seems to be a problem with the localizer, as I have the same error when trying the localization video as well. I'm running on master commit 861b037403a3e3.

kitsukawa, 2016-10-10:
How did you build the programs? Can you build using `./catkin_make_release` instead of executing `catkin_make -DCMAKE_BUILD_TYPE=Release`.

WeirdAI, 2016-10-13:
Same as me, the ndt-matching process died. I'm not sure it is caued by the TF error or it causes the TF error, there is no TF tranform from Velodyne to the World. Have you solved your problem?

wgcv, 2016-10-13:
I download the sample maps and bag, but the launch where i can download? I dont see nothing when i run

weisongwen, 2017-05-16:
hi, kitsukawa, I get the same problem:
```
ndt_matching: /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:430: static void Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::run(Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, const Lhs&, const Rhs&, Packet&) [with Lhs = Eigen::Block<const Eigen::Matrix<double, 6, 6>, 6, -1, true>; Rhs = Eigen::Matrix<double, -1, 1, 0, 6, 1>; Packet = __vector(2) double; int LoadMode = 1; Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index = long int]: Assertion `lhs.cols()>0 && "you are using a non initialized matrix"' failed.
[ndt_matching-6] process has died [pid 2797, exit code -6, cmd /home/foo/catkin_ws/devel/lib/ndt_localizer/ndt_matching __name:=ndt_matching __log:=/home/foo/.ros/log/d03f04ce-8eb0-11e6-9c56-408d5cb83aa0/ndt_matching-6.log].
log file: /home/foo/.ros/log/d03f04ce-8eb0-11e6-9c56-408d5cb83aa0/ndt_matching-6*.log

```
My system is: Ubuntu14.04 +indigo
I compile the program build using **./catkin_make_release**, 

Thanks!

kitsukawa, 2017-05-16:
Can you follow the steps in the video below instead of using Quick Start tab? Quicks Start tab may not keep up with the current version of ndt_matching.
https://www.youtube.com/watch?v=ODlxMzGTJzw

weisongwen, 2017-05-17:
Thanks for your excellent work and  your reply!  I do as [the https://www.youtube.com/watch?v=ODlxMzGTJzw](url)

But I still get this `Error:ndt_matching: /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:430: static void Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::run(Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, const Lhs&, const Rhs&, Packet&) [with Lhs = Eigen::Block<const Eigen::Matrix<double, 6, 6>, 6, -1, true>; Rhs = Eigen::Matrix<double, -1, 1, 0, 6, 1>; Packet = __vector(2) double; int LoadMode = 1; Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index = long int]: Assertion `lhs.cols()>0 && "you are using a non initialized matrix"' failed.
[ndt_matching-6] process has died [pid 2797, exit code -6, cmd /home/wenws/catkin_ws/devel/lib/ndt_localizer/ndt_matching __name:=ndt_matching __log:=/home/wenws/.ros/log/d03f04ce-8eb0-11e6-9c56-408d5cb83aa0/ndt_matching-6.log].
log file: /home/wenws/.ros/log/d03f04ce-8eb0-11e6-9c56-408d5cb83aa0/ndt_matching-6*.log`

weisongwen, 2017-05-17:
it says that `you are using a non initialized matrix"' failed`

On RViz, I  click "2D Pose Estimate" and put an arrow on the map where there is the vehicle. but this still did not work.  do you have any other idea about this bug..

Thanks

weisongwen, 2017-05-17:
I find that when I start the ndt_matching node, one of my CPU(my computer have 8 cpu) run to 100% in terms of the usage. should this be the reason resulting the Error :`Error:ndt_matching: /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:430: static void Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::run(Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, const Lhs&, const Rhs&, Packet&) [with Lhs = Eigen::Block<const Eigen::Matrix<double, 6, 6>, 6, -1, true>; Rhs = Eigen::Matrix<double, -1, 1, 0, 6, 1>; Packet = __vector(2) double; int LoadMode = 1; Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index = long int]: Assertionlhs.cols()>0 && "you are using a non initialized matrix"' failed.`  ?

My computer parameters are:
Memory:7.8 GiB
processor:Intel® Core™ i7-6700HQ CPU @ 2.60GHz × 8 
graphics:GeForce GTX 1060/PCIe/SSE2
operation type:64-bit Ubuntu 14.04

Thanks !

kitsukawa, 2017-05-18:
I have no idea on that error. Can you record and share the screen capture of the operation?

weisongwen, 2017-05-18:
[image]

[image]

[image]

[image]

[image]

[image]

[image]
this is the  recorded  screen capture of the operation

kitsukawa, 2017-05-18:
As I told the day before yesterday, can you exactly follow the steps in the video below instead of using Quick Start tab? In the video, Quick Start tab is never used.
https://www.youtube.com/watch?v=ODlxMzGTJzw

This is the copy&paste of the caption of the video.
This video demonstrates how to localize the vehicle using rosbag data.
1. Go to Simulation tab and select a rosbag which includes /points_raw.
2. Click "Play" and click "Pause" to set rosparam "use_sim_time" true.
3. Go to Setup tab, select Localizer (Velodyne or Hokuyo 3D URG), input parameters related to relative position between the vehicle and localizer, push "TF" and push "Vehicle Model" (if you leave the space blank, the default model will be loaded.).
4. Go to Map tab and load pointcloud map and publish TF between /world and /map. (Refer to https://www.youtube.com/watch?v=EJa4P...)
5. Go to Sensing tab and run voxel_grid_filter to downsample scan data.
6. Run ndt_mathcing, click "app" and choose Initial Pos.
7. Go to Simulation tab and click "Pause" to replay rosbag.
8. On RViz, click "2D Pose Estimate" and put an arrow on the map where there is the vehicle.

Note that RViz config is /Autoware/ros/src/.config/default.rviz.

weisongwen, 2017-05-20:
Hi, @kitsukawa ,so glad to get your reply!
Thanks ,Let me try again.

weisongwen, 2017-05-26:
Hi, @chiyuan-goh , I did it follow the process that you advised:

https://www.youtube.com/watch?v=ODlxMzGTJzw
This is the copy&paste of the caption of the video.
This video demonstrates how to localize the vehicle using rosbag data.

`Go to Simulation tab and select a rosbag which includes /points_raw.
    Click "Play" and click "Pause" to set rosparam "use_sim_time" true.
    Go to Setup tab, select Localizer (Velodyne or Hokuyo 3D URG), input parameters related to relative position between the vehicle and localizer, push "TF" and push "Vehicle Model" (if you leave the space blank, the default model will be loaded.).
    Go to Map tab and load pointcloud map and publish TF between /world and /map. (Refer to https://www.youtube.com/watch?v=EJa4P...)
    Go to Sensing tab and run voxel_grid_filter to downsample scan data.
    Run ndt_mathcing, click "app" and choose Initial Pos.
    Go to Simulation tab and click "Pause" to replay rosbag.
'

sometimes, I can successfully replicate the localization process in the video, but sometimes failed with the previous error:

**_Error:ndt_matching: /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:430: static void Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::run(Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, const Lhs&, const Rhs&, Packet&) [with Lhs = Eigen::Block<const Eigen::Matrix<double, 6, 6>, 6, -1, true>; Rhs = Eigen::Matrix<double, -1, 1, 0, 6, 1>; Packet = __vector(2) double; int LoadMode = 1; Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index = long int]: Assertionlhs.cols()>0 && "you are using a non initialized matrix"' failed._**

I checked the [requirements](https://github.com/weisongwen/Autoware) about the computer :
**Space Recommendation
    of CPU cores: 8
    RAM size: 32GB
    Storage size: 30GB**

**But my computer's  RAM is 8G, the swap space of my ubuntu system is 7.8G .** Can this be the issue resulted in the Error?

Thanks!

ZuoJiaxing, 2017-06-17:
@weisongwen, Hi, I also meet the same problem with you  when I  use the pcl ndt-matching:

Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::run(Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, const Lhs&, const Rhs&, Packet&) [with Lhs = Eigen::Block<const Eigen::Matrix<double, 6, 6>, 6, -1, true>; Rhs = Eigen::Matrix<double, -1, 1, 0, 6, 1>; Packet = __vector(2) double; int LoadMode = 1; Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index = long int]: Assertion `lhs.cols()>0 && "you are using a non initialized matrix"' failed.

And when it falls, one of my CPU(my computer have 8 cpu) also run to 100%. So how to fix this proble alout the NDT of PCL? Thanks a lot!

weisongwen, 2017-06-17:
Hi, @kitsukawa @ZuoJiaxing , I still did not handle this problem, though I do as what you told me to do..

 Hi, @kitsukawa  I did it follow the process that you advised:

https://www.youtube.com/watch?v=ODlxMzGTJzw
This is the copy&paste of the caption of the video.
This video demonstrates how to localize the vehicle using rosbag data.

`Go to Simulation tab and select a rosbag which includes /points_raw.
Click "Play" and click "Pause" to set rosparam "use_sim_time" true.
Go to Setup tab, select Localizer (Velodyne or Hokuyo 3D URG), input parameters related to relative position between the vehicle and localizer, push "TF" and push "Vehicle Model" (if you leave the space blank, the default model will be loaded.).
Go to Map tab and load pointcloud map and publish TF between /world and /map. (Refer to https://www.youtube.com/watch?v=EJa4P...)
Go to Sensing tab and run voxel_grid_filter to downsample scan data.
Run ndt_mathcing, click "app" and choose Initial Pos.
Go to Simulation tab and click "Pause" to replay rosbag.
'

sometimes, I can successfully replicate the localization process in the video(one success in 100 trail), but always failed with the previous error:

Error:ndt_matching: /usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:430: static void Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::run(Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index, const Lhs&, const Rhs&, Packet&) [with Lhs = Eigen::Block<const Eigen::Matrix<double, 6, 6>, 6, -1, true>; Rhs = Eigen::Matrix<double, -1, 1, 0, 6, 1>; Packet = __vector(2) double; int LoadMode = 1; Eigen::internal::product_packet_impl<0, -1, Lhs, Rhs, Packet, LoadMode>::Index = long int]: Assertionlhs.cols()>0 && "you are using a non initialized matrix"' failed.

I checked the requirements about the computer :
Space Recommendation
of CPU cores: 8
RAM size: 32GB
Storage size: 30GB

But my computer's RAM is 8G, the swap space of my ubuntu system is 7.8G . Can this be the issue resulted in the Error?

Thanks!

kitsukawa, 2017-06-18:
By the way, what data are you using? Sample Moriyama data which can be downloaded from Autoware Github page?

weisongwen, 2017-06-18:
Hi, @kitsukawa , Thanks for your reply!

 I use the data downloaded from Autoware Github page and do as what you told me to do. I still get the problem. 

could this be a problem from tf ? actually, I have launched the tf.launch file..

weisongwen, 2017-06-22:
@ZuoJiaxing , Hi

Have you solved this problem? 

I tried many ways but still failed.

pxlong, 2017-08-09:
@ZuoJiaxing @weisongwen I've successfully replicated the quick start demo today:
you can try to update PCL to the latest version (1.7.2) first, 
` sudo add-apt-repository ppa:v-launchpad-jochen-sprickerhof-de/pcl
  sudo apt-get update
  sudo apt-get dist-upgrade`
then run ./catkin_make_release and follow the instructions in Autoware_QuickStart_v1.1.pdf

weisongwen, 2017-08-13:
Hi, @pxlong 
Thanks for your reply and way to that issue. 

Best regards

weisongwen, 2017-08-13:
Hi, @kitsukawa 

I also handle the problem by reinstalling the ros with jade version. 
the issue mentioned  may be result in what @pxlong  said above.

Best regards

weisongwen, 2018-06-29:
nice to hear such goog news that you solve the probelm,
best 
welson

kfunaoka, 2018-07-13:
Solved?

kfunaoka, 2019-05-16:
We can go to [ROS Answers](https://answers.ros.org/questions/) if some problems remain.

closed 2019-05-16 by kfunaoka