autowarefoundation/autoware #472: way point follower strange behavior?

Benjaminmar8, 2016-10-10:
Dear all, I am testing the way point follower by following the "planning with rosbag" tutorial. I have a recorded bag file and a self-generated map. Everything looks ok to me except the pure pursuit of way point follower has a strange behavior, the path seems like being rotated by 90 degrees. Here is a picture illustrating the problem: 
[image]
My base_link to localizer static tf setting is: 0, -1.04, 1.71, 0, 0, 0
I have tried to change the above setting to: 0, -1.04, 1.71, 1.571, 0, 0 and 0, -1.04, 1.71, -1.571, 0, 0. However, once I changed the yaw to +/- 1.571, ndt_matching failed immediately and the base_link of the car "flyes" randomly in the global map. Could you help me with this issue? Thank you very much.

Benjaminmar8, 2016-10-10:
Solved. Cause: One set of static tf is used to generate the map, another set of static tf is used to record way points.

closed 2016-10-10 by Benjaminmar8

sgzzgit, 2016-11-01:
Sorry to bother you.
Could you tell me how to show the speed information in rviz as in the up and left side of your Picture.
Thanks very much. @Benjaminmar8

Benjaminmar8, 2016-11-01:
@sgzzgit default rviz setting, km/h display type, linear_velocity topic

bbidault2, 2017-01-31:
Hi @Benjaminmar8, I have a very similar problem to yours and I don't fully understand your solution.
> One set of static tf is used to generate the map
You don't need to set a static tf from base_link to localizer to generate a map, what are you making reference to ? Could you give more details, thank you.

Benjaminmar8, 2017-02-01:
the tf must be consistent. In my case, tfs between base_link and localizer are different between "recorded bag file" and off-line sim later on.