autowarefoundation/autoware #479: Improving large-scale ndt mapping performance

Benjaminmar8, 2016-10-14:
Dear All, 
Referring to this issue #457 , the large-scale map I build has an error around 5-6 meters after 1.5 km driving. My question is, what would be the approach to improve ndt-mapping performance, especially for large-scale maps? One approach that I am thinking is to use IMU and GNSS (with RTK) equipments to aid the ndt-mapping. Is that feasible? Thanks.

sgzzgit, 2016-10-14:
loop closure may improve the accuracy of ndt_mapping.

Carrie0416, 2016-12-13:
Could you tell me how to get the source code with loop closure for ndt_mapping?

Carrie0416, 2016-12-16:
When I use the ndt_mapping, I have the same problem for this error. Cartographer_ros's point cloud2 (scan_matched_points2) can't use in the ndt_matching. Could you please give me some advice?

wallizadaM, 2017-03-13:
I am also very interested in a solution for this issue. In addition to x, y, i get huge errors in z axis specially on turns.

elberthuang, 2017-06-05:
I got the same problem. Anybody share some experience of solving it? It could be a fatal problem when doing large-scale mapping. Thanks!

kfunaoka, 2018-07-13:
sgzzgit's answer seems to be the current final answer.

closed 2018-07-13 by kfunaoka