autowarefoundation/autoware #487: Localization accuracy using ndt matching at turns

nsakar, 2016-10-28:
I am having some localization accuracy issues at turns. As I understand the convergence loop uses the current calculated pose and velocity as the initial guess for the next iteration till the convergence and scan matching occurs. Are there other odometry data that can be used as the initial guess for the iteration that may make the calculations more accurate? For example canbus actual linear and angular velocity?

yukkysaito, 2017-01-29:
@kitsukawa Could you please reply?

kitsukawa, 2018-02-20:
@nsakar Now, ndt_matching supports imu and odometry in develop branch, and calculate initial guess for the iteration using imu and odemetry data. We tested and found that combination of imu or/and odometry improves the accuracy to a certain degree.

closed 2018-02-20 by kitsukawa