autowarefoundation/autoware #492: gps coordinate conversion from llh to xyz

nsakar, 2016-10-30:
Hi 
My vehicle returns gps/fix values in lat, lon, altitude format [sensor_msgs/NavSatFix]and I need to convert it to x,y, z coordinates. I was studying the fix2tfpose node and the geo_pos_conv library that it uses.
I have the following questions:
1. What does the set_plane function do? In the USA what should this value be set to?
2. Is the NavSatFix values [lat,lon, altitude] converted to EKEF format by the conv_llh2xyz method in the library? Is there some documentation I can reference? The x, y, z coordinates are returned against what datum? Is there a correlation to the USA state plane system for the projection?

kitsukawa, 2016-11-01:
> What does the set_plane function do? In the USA what should this value be set to?
The set_plane function sets the plane number and sets parameters which depends on the plane. In Japanese rectangular plane coordinate system, there are 19 planes, in other words, there are 19 reference points according to the regions. I think this function is invalid in the other coordinate systems.

> Is the NavSatFix values [lat,lon, altitude] converted to ECEF format by the conv_llh2xyz method in the library? Is there some documentation I can reference? The x, y, z coordinates are returned against what datum? Is there a correlation to the USA state plane system for the projection?
No, conv_llh2xyz is for Japanese coordinate system mentioned above. You have to implement functions to convert lat, lon, altitude to x,y,z in ECEF format.

closed 2016-11-01 by nsakar

thomas3016, 2017-08-17:
@nsakar Have you salve this problem

dejanpan, 2017-08-25:
Use https://geographiclib.sourceforge.io/html/classGeographicLib_1_1Geocentric.html for llh to ECEF conversion. Then you can set one ECEF point as your origin and compute offset which will be your local cartesian coordinates.

jokla, 2018-04-27:
Hi all!

If I want my local frame in 45°46'N 4°43'E,  Could I just add in the file [geo_pos_conv.cpp](https://github.com/CPFL/Autoware/blob/master/ros/src/computing/perception/localization/lib/gnss/src/geo_pos_conv.cpp) the following plane?:

` ` ` 
 else if (num == 20)
  {
    lon_deg = 45;
    lon_min = 46;
    lat_deg = 4;
    lat_min = 43;
}
` ` `

Or do I need to create a new conversion using the library mentioned by @dejanpan?

wsung1, 2018-05-23:
Hi, @jokla, by any chance, have you got the answer to your question?

ahuer2435, 2018-05-24:
Maybe you  can refer：
https://github.com/ros-drivers/gps_umd/commit/a80ea0c6695f50261bf5c5bb1b1bc4173f179aa4
line148 convert wgs84 to utm.

jokla, 2018-05-24:
Hi @wsung1 , I have not got the answer but I have just added a new plane with my desired local frame and it was working fine. It is not a clean solution but I read somewhere that they are working on a new way to define the local frame.

wsung1, 2018-05-24:
Thanks @jokla and @ahuer2435, I'll give it a try  and let you guys know its result.

javcursor, 2018-09-07:
Hi everybody, I'm trying use the suggested library but I'm in a blocking point. I've modified the plane 19 with my world origin coordinates. I have changed the geo_pos_conv::conv_llh2xyz(void):

`void geo_pos_conv::conv_llh2xyz(void)
{
const GeographicLib::Geocentric& earth = GeaographicLib::Geocentric::WGS84();
GeographicLib::LocalCartesian proj(lat_map_origin, lon_map_origin, 0, earth);
proj.Forward(m_lat_degrees, m_lon_degrees, 0, m_x, m_y, m_z); //We use GPRMC, so m_lat from set_llh_nmea_degrees is radians, convert to degrees for library
}`

In RViz I can see the world and map origin in the correct place but baselink and gps no. Are correct the changes that I've made? Do I need modify something more?

Thanks

javcursor, 2018-09-12:
Hi, the conversion has coherent values of X,Y,Z after check the sign of latitude and longitude. I've added a static transform between base_link and gps because they was separated. 
I let a video with the actual behavior, the dataset and my src folder with modifications:
- geo_pos_conv.cpp
- nmea2tfpose_core.cpp
- ndt_matching.launch

I'm using Ubuntu 16.04, ROS Kinetic and Autoware 1.8, any help is welcome.
@kitsukawa @wsung1