autowarefoundation/autoware #503: Can we run autoware on ROS kinetic?

YAMMAY, 2016-11-07:
Can we run autoware on ROS kinetic?

reinzor, 2017-01-16:
ping

yukkysaito, 2017-01-29:
No.
The dependent library does not support kinetic.

However, we are working toward that.
If you want to run on kinetic, you have to use docker now.

anelliot, 2017-03-07:
Is there any update on this? Which library dependencies specifically are blocking you? If this is in progress, on which branch are you doing this work?

HolmarSeffelgruber, 2017-03-31:
@yukkysaito Do I understand correctly that I can use autoware with kinetic using docker? How to do that?

yukkysaito, 2017-03-31:
Now we update autoware in feature/kinetic branch. But some package cannot work.

HolmarSeffelgruber, 2017-03-31:
So the package can't be compiled yet?
I tried to follow the instructions for Ubuntu 16.04 but catkin make fails (e.g. I need ROS indigo or jade for some packages. See the txt file)

[catkin_make_release_output.txt](https://github.com/CPFL/Autoware/files/884855/catkin_make_release_output.txt)

yukkysaito, 2017-03-31:
@HolmarSeffelgruber did you compile in feature/kinetic?

HolmarSeffelgruber, 2017-03-31:
@yukkysaito right I forgot to add "--branch feature/kinetic" in the cloning process. Thanks!

yukkysaito, 2017-03-31:
@HolmarSeffelgruber Pls run following command in Autoware dir.
% git checkout feature/kinetic
% cd ./ros
% ./catkin_make_release

yukkysaito, 2017-03-31:
https://autoware.herokuapp.com/
This is autoware slack team.
Please join, if it's OK with you.

HolmarSeffelgruber, 2017-03-31:
@yukkysaito the checkout responds that it's up to date already. I think "git clone https://github.com/CPFL/Autoware.git --branch feature/kinetic" worked fine.
Do you mean joining the developement team? I don't think I got much time for it. I'm preparing to write my bachelor thesis with a project right now.

yukkysaito, 2017-03-31:
This is open slack team. Anyone can Q&A. If you wanna quick response, you should join

HolmarSeffelgruber, 2017-03-31:
Thanks! Althought I don't think it can be much quicker than you responded here!

closed 2017-05-01 by yk-fujii