autowarefoundation/autoware #505: TF ERROR

YAMMAY, 2016-11-09:
No transform from [base_link] to [world]
No transform from [gps] to [world]
No transform from [velodyne] to [world]

yukkysaito, 2017-01-29:
I think it is caused by simulation clock.
If you use rosbag, pls play rosbag in order to enable simulation clock at first.

closed 2017-05-29 by yk-fujii

elberthuang, 2017-06-02:
@yukkysaito Hello! What is the sequence? Enable "use_sim_time" first or launch rviz first? What about TF? I am really confused. Sometimes I get correct results when running ndt matching, sometimes don't. Can you explain the correct sequence when running ndt mathcing from the beginning?

kitsukawa, 2017-06-02:
This is the corret sequence to run ndt_matching.
https://www.youtube.com/watch?v=5bj7gkFlul0
1. Go to Simulation tab and select a rosbag which includes /points_raw and /nmea_sentence(or /gnss_pose).
2. Click "Play" and click "Pause" to set rosparam "use_sim_time" true.
3. Go to Setup tab, select Localizer (Velodyne or Hokuyo 3D URG), input parameters related to relative position between the vehicle and localizer, push "TF" and push "Vehicle Model" (if you leave the space blank, the default model will be loaded.).
4. Go to Map tab and load pointcloud map and publish TF between /world and /map. (Refer to https://www.youtube.com/watch?v=EJa4P...)
5. Go to Sensing tab and run voxel_grid_filter to downsample scan data.
6. Go to Computing tab and run nmea2tfpose to convert /nmea_sentence into /gnss_pose.
7. Run ndt_mathcing, click "app" and choose GNSS as initial position.
8. Go to Simulation tab and click "Pause" to replay rosbag.

elberthuang, 2017-06-03:
@kitsukawa Thanks a lot!