autowarefoundation/autoware #509: Autoware build fails

natalielucker, 2016-11-15:
Hi there

I followed the instructions of building Autoware on Ubuntu 15.04, but encountered the following error
```
Linking CXX shared library /home/natalie/Autoware/ros/devel/lib/libintegrated_viewer.so
[ 68%] Built target integrated_viewer
Linking CXX shared library /home/natalie/Autoware/ros/devel/lib/libvector_map.so
[ 68%] Built target vector_map
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```
One of the suggestions was to build this library separately and then try to run `./catkin_make_release` again, but it didn't help - error appeared in another place of installation. 
What should I do?

amc-nu, 2016-11-15:
Hello Nat,
Don't you see any other error in the terminal?

natalielucker, 2016-11-17:
I can provide you with log-files.
[Logs.zip](https://github.com/CPFL/Autoware/files/596714/Logs.zip)

amc-nu, 2016-11-18:
Can you check if you have installed the tf package from ros?
http://wiki.ros.org/tf

natalielucker, 2016-11-28:
Yes, tf package is installed
```
~$ rospack find tf
/opt/ros/jade/share/tf
```

amc-nu, 2016-11-28:
Nat, 
I see you're not using indigo.
For the time being Autoware is only compatible with Indigo.
Best regardsAbraham

natalielucker, 2016-11-28:
Abraham,
In official requirements it is said, that both ROS indigo(Ubuntu 14.04) and ROS jade(Ubuntu 15.04) are maintained.

amc-nu, 2016-11-28:
Nat,
Sorry my bad,you're on Ubuntu 15
I didn't know that.

Then the problem is evident. The developer of OrbSlam hard coded the include of TF to indigo directory, as shown in the log your kindly attached.
KeyFrame.cc

If you're in a hurry please change the path of the include to jade. We'll ask @sujiwo to change it.

That should solve your problem.

closed 2016-12-01 by amc-nu