autowarefoundation/autoware #514: How to load  Camera or some drivers?

Fuji-ken, 2016-11-17:
I have pointgrey's Camera.
I've once suceeded to draw a Camera's view in Ubuntu by Flycapture.

So I tried to rebuild cameradrivers, so that it can start from Autoware-GUI.
But I can't see which holder is driver's one  I have to install to. 
I challenged log is as bellow.
"~Autoware/ros/src/$ git https://github.com/ros-drivers/pointgrey_camera_driver.git"
"~Autoware/ros/src/$cd ../"
" ~Autoware/ros/$ catkin_make"
Then, "Autoware/ros/src/pointgrey_camera_driver" is created.
After that, If I try to run "Autoware/ros/.run", that's GUI creation hour been to very long.
What's wrong settings did I do?
Moreover, Camera's View process on FlyCapture also been to very slowry.
Something wrong?

If anyone has solutions, please teach me!
Regs,

amc-nu, 2016-11-20:
Hi Ken

Can you tell us:
1. which pointgrey camera are you using?
2. Is it USB or Ethernet?
3. How are you launching the camera node?

amc-nu, 2016-11-21:
> Hi AM
>
> My Camera is as below.
> Product Family:Grasshopper3
> CameraModel:GS3-U3-28S4C-C
> Connector:USB
>
> I can't see I have to create a program by C++ or Python which can detect  the camera port.
> The word "node" you called is this program?
> I don't stil create aprogram for seeing camera port.
> At first, I think pointgrey's library I download from below is included all necesarry "node"program.
> •Source: git https://github.com/ros-drivers/pointgrey_camera_driver.git
>
> Is it mistake?
>
> If so, the camera is'nt exist on Autoware's GUI.
> Basicaly, I can't see how to use Autoware's Usage.
>
> Could you show me how to create the Autoware environment for using Camera or Laser facility, please?
>
> Regs, Ken

Ken-san
We are not using the ROS drivers provided for pointgrey. We developed our own ROS node( ROS program). 
You don't need to develop a new program. It is contained in Autoware.
1. Can you see the camera stream using the FlyCapture2 program from pointgrey?
2. If you can do it, then just launch Autoware.
If you can't, make sure you followed the setup instructions given by pointgrey to configure the USB 3.0
3. Go to the sensing tab, click the point grey and you should be able to view the camera stream with Rviz.

Fuji-ken, 2016-11-21:
Hi, AM
Thank you for your advice.
I'll try to take it later.
Now, I have another questions.
From this morning, I tried to view a camera display the following story.
Then, I  can achieve to display the view, but it's color is grey or green.
Do you know how to set the view to real colors?
I think I can set the following prcess as a launch-file and set to Autoware procedure,
I  could set the view in Autoware environment.

Is it a unnecessary process?

$ sudo apt-get install ros-indigo-pointgrey-camera-driver
$ rosrun pointgrey_camera_driver list_cameras
$ roslaunch pointgrey_camera_driver camera.launch
$ roslaunch pointgrey_camera_driver camera.launch camera_serial:=*******
$ rosrun image_view image_view image:=/camera/image_raw
[INFO][1479692347.486413783]:Using transport "raw"

Please show me your advice.

Regs

amc-nu, 2016-11-21:
Ken,
There is no need to install the ROS drivers. Once you installed Autoware and you are able to see the camera stream with FlyCapture2, you need to run:
`roslaunch pointgrey grasshopper3.launch`
then you'll be able to view it in RViz with the `/image_raw` topic.

closed 2016-12-01 by amc-nu

Fuji-ken, 2016-12-01:
Thanks so much for teaching me many advice.
I've been to clear above problem at last.
Best regards