autowarefoundation/autoware #533: Stitching the .pcd mapps

wallizadaM, 2016-12-03:
Hi, I have large rosbag file, which i have divided into smaller rosbag files. Each rosbag file is then used to create a .pcd by running ndt mapping. Now each .pcd has origin at 000 and i get the end position and orienation based on the  ndt mapping. 

My question is how to stich this local maps into a global map. This issue was discussed in Issue #428. However the suggestion by Benjaminmar8 commented on Aug 26  does not work.

For example, if we have 1.pcd,2.pcd, 3.pcd. and assuming we have end coordinates for 1.pcd (acquired from ndt mapping) is 3 4 5 0.1 0.001 0.057, we do the following:
rosrun ndt_localizer local2global ~/2.pcd 3 4 5 0.1 0.001 0.057
And then, use Autoware to load the map 1.pcd. local2gloabl node will generate a "global_map.pcd", which is the transformed version of 2.pcd.

Now the question is, although we have a position and oriantation locally for 2.pcd from ndt mapping similar to 1.pcd, how to get the end position and orientation of global_map.pcd (gloabl 2.pcd) so it can be used for transformation of 3.pcd.

kitsukawa, 2016-12-04:
How about using ndt_mapping in large_scale_ndt_mappig branch. I modified ndt_mapping so that it outputs pcd files every 200m (You can change this value in ndt_mapping.launch) in .ros directory. You don't have to register each pcd file.

wallizadaM, 2016-12-04:
Thanks . I used the ndt_mapping in large_scale_ndt_mapping branch. The way i used it was start simulation of the large .bag file, pauss it at 0%, set the velodyne to baselink transformation in Setup tab, then select ndt_mapping in the Computing tab, start the simulation. Two issues: First, was expecting pcd files to be generated as it goes, but no pcd file was generatd. When simulation finished, i saved pcd file using ndt_mapping app, but that only saved the last part of the map. Second issue,  the map generated gets very erroneous in the z axis either going very high or very low. This was observed in the master branch aswell. Could you please advise.

wallizadaM, 2016-12-04:
Is there a way to set Z to zero, since i am mapping a plane environment with no up and downs.

kitsukawa, 2016-12-04:
Regarding the first issue, did you check the .ros in the home directory? It's not the Autoware/ros directory. As for second issue, I guess mapping error in the z axis happens if the velocity of the vehicle when recording the rosbag is fast. Can you try to do mapping using another rosbag recorded in a low velocity if possible. Or using more brute force method, set current_pose.z = 0.0 after matching in the source code.

wallizadaM, 2016-12-08:
Thanks. First issue sorted. In regards to second issue, tried mapping with car running at speed of 5KMH and 10KMH, still get error in z axis, also the x and y axes gets very off. Trying to map a large container yard area of nearly to 2KMs with loops. 

Any advise on the strategy of data collection for mapping. 
Currently have tried collecting data for mapping starting from start location, doing the loops and then coming back to start location. Also have tried collecting data in segments and then stitching them.

Note: I am using velodyne VLP16 with calibration file that comes with autoware.

bbidault2, 2017-01-27:
Hi all, first, thank you for your contribution.
I use a Velodyne 32 to do mapping and localization and I am now interested in large scale mapping.
@kitsukawa in your first answer, you mention a ndt_mapping node in a branch named large_scale_ndt_mapping that output a .pcd file every 200 meters. I would be very interested in such a feature but I cannot find it. Same, I cannot find any reference to this node in ndt_mapping.launch.
Can you help me with that, thank you.

bbidault2, 2017-02-07:
I ended up solving my problem by modifying ndt_mapping.cpp, you can check out my fork here -> https://github.com/bbidault2/Autoware/blob/my-autoware/ros/src/computing/perception/localization/packages/ndt_localizer/nodes/ndt_mapping/ndt_mapping.cpp

I believe we can close the issue now.

dlounis, 2018-01-12:
Hi,

I am trying to build an entire map from a kitti dataset. I am wondering if the component that deals with large maps (automatically saves PCD sub-maps every N meters) is already included in the version of Autoware that I have installed recently (from Master branch).

Regards.

kitsukawa, 2018-01-15:
@dlounis You can use approximate_ndt_mapping. It output PCD submaps every N meters. You can change N from app dialog next to approximate_ndt_mapping in Computing tab.

kitsukawa, 2018-01-15:
@dlounis Yes. submap_size.

dlounis, 2018-01-15:
@kitsukawa what does this size correspond to ??? lengths ? area ? what are the units ? meters ? square meters ?

kitsukawa, 2018-01-16:
@dlounis It's distance. The units are meters.

likui01, 2018-02-28:
@kitsukawa  how to combine all the submaps into one map. meaning one PCD file.?

kitsukawa, 2018-03-01:
@likui01 You want to make one PCD file containing all the points of several PCDs? I don't know why you want to do this, but I think you can do this using [pcl_ros](http://wiki.ros.org/pcl_ros)  pointcloud_to_pcd. First publish all the PCDs using points_map_loader in Map tab of Runtime Manager. After that, save the all the published pointcloud to a PCD using pointcloud_to_pcd in pcl_ros.

likui01, 2018-03-01:
@kitsukawa  i used approximate ndt mapping to generate the map. it save the map into 12 submap files. 
so i am wondering how to combine all of them . there local frame work could be combine in global

kitsukawa, 2018-03-01:
@likui01 points_map_loader in Map tab of Runtime Manager support loading multiple PCDs and it combines multiple PCDs and publishes as one massage in /points_map topic. So you don't need to combine the PCDs into one file.

likui01, 2018-03-02:
@kitsukawa  when i try to load multiple files
i get this error and no map is loaded
[ERROR] [1519951824.063989387]: a message of over a gigabyte was predicted in tcpros. that seems highly unlikely, so I'll assume protocol synchronization is lost.

kitsukawa, 2018-03-02:
The maps are very dense so you need to downsample the pointcloud. You can downsample PCDs using `PCD Filter` in Map tab.

halderb, 2019-07-03:
Hi, I am having a problem with combining multiple pcd files into a global pcd file. My main aim is to create offline map for my university. In the beginning, I collected data and stored it in multiple separate rosbag files which I have converted into pcd files by performing ndt mapping using Autoware. When I try to merge the pcd files by uploading them for Mapping, I get an overlapped maps in rviz. I tried "rosrun ndt_localizer local2global" command for manually stitching the pcd files but it didn't work as my Autoware software doesn't have ndt_localizer package. Can you tell me a way out of how to combine my pcd files into a global pcd file?

Pallav1299, 2020-03-16:
There seems to be some translation between the end and start points of two consecutive PCD's generated using approximate_ndt_mapping. Plus, they are also processing points and generating output inside the points_callback. Is it really ok to do so?