autowarefoundation/autoware #552: sample_points_map error when compling

xpharry, 2016-12-31:
make[2]: make[2]: *** [/home/peng/Autoware/ros/devel/lib/sample_data/sample_points_map] Error 1
*** [/home/peng/Autoware/ros/devel/lib/sample_data/sample_points_map_update] Error 1
make[1]: *** [util/packages/sample_data/CMakeFiles/sample_points_map.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [util/packages/sample_data/CMakeFiles/sample_points_map_update.dir/all] Error 2

xpharry, 2016-12-31:
also
/usr/bin/ld: cannot find -lDummyDevice
/usr/bin/ld: cannot find -lOniFile
/usr/bin/ld: cannot find -lPS1080

It seems that it needs support from OpenNI2, which I have installed.

kitsukawa, 2017-01-04:
Execute the following command before compiling.
$ export LIBRARY_PATH=/usr/lib/OpenNI2/Drivers:$LIBRARY_PATH

closed 2017-06-12 by yk-fujii