autowarefoundation/autoware #553: orb_localizer compiling error

xpharry, 2017-01-04:
/home/peng/Autoware/ros/devel/lib/liborb_slam2.a(Map.cc.o): In function `boost::archive::detail::common_iarchive<boost::archive::binary_iarchive>::vload(boost::archive::class_name_type&)':
Map.cc:(.text._ZN5boost7archive6detail15common_iarchiveINS0_15binary_iarchiveEE5vloadERNS0_15class_name_typeE[_ZN5boost7archive6detail15common_iarchiveINS0_15binary_iarchiveEE5vloadERNS0_15class_name_typeE]+0x1): undefined reference to `boost::archive::basic_binary_iarchive<boost::archive::binary_iarchive>::load_override(boost::archive::class_name_type&)'
/home/peng/Autoware/ros/devel/lib/liborb_slam2.a(Map.cc.o): In function `void boost::serialization::throw_exception<boost::archive::archive_exception>(boost::archive::archive_exception const&)':
Map.cc:(.text._ZN5boost13serialization15throw_exceptionINS_7archive17archive_exceptionEEEvRKT_[_ZN5boost13serialization15throw_exceptionINS_7archive17archive_exceptionEEEvRKT_]+0x1d): undefined reference to `boost::archive::archive_exception::archive_exception(boost::archive::archive_exception const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/peng/Autoware/ros/devel/lib/orb_localizer/orb_matching] Error 1
make[1]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/all] Error 2
/home/peng/Autoware/ros/devel/lib/liborb_slam2.a(Map.cc.o): In function `boost::archive::detail::common_iarchive<boost::archive::binary_iarchive>::vload(boost::archive::class_name_type&)':
Map.cc:(.text._ZN5boost7archive6detail15common_iarchiveINS0_15binary_iarchiveEE5vloadERNS0_15class_name_typeE[_ZN5boost7archive6detail15common_iarchiveINS0_15binary_iarchiveEE5vloadERNS0_15class_name_typeE]+0x1): undefined reference to `boost::archive::basic_binary_iarchive<boost::archive::binary_iarchive>::load_override(boost::archive::class_name_type&)'
/home/peng/Autoware/ros/devel/lib/liborb_slam2.a(Map.cc.o): In function `void boost::serialization::throw_exception<boost::archive::archive_exception>(boost::archive::archive_exception const&)':
Map.cc:(.text._ZN5boost13serialization15throw_exceptionINS_7archive17archive_exceptionEEEvRKT_[_ZN5boost13serialization15throw_exceptionINS_7archive17archive_exceptionEEEvRKT_]+0x1d): undefined reference to `boost::archive::archive_exception::archive_exception(boost::archive::archive_exception const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/peng/Autoware/ros/devel/lib/orb_localizer/orb_mapping] Error 1
make[1]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_mapping.dir/all] Error 2

yukkysaito, 2017-01-29:
I have never seen that error.
Please tell me the environment.

yukkysaito, 2017-01-29:
@hatem-darweesh do you know this error?

sujiwo, 2017-01-29:
Please install libboost-serialization-dev

xpharry, 2017-01-30:
@sujiwo This one has been installed.

xpharry, 2017-01-30:
@yukkysaito 
utuntu 14.04 + ros indigo
Dell XPS 9600

sujiwo, 2017-01-31:
I'm trying to address this bug from new branch (orb_slam_tc).

xpharry, 2017-01-31:
@sujiwo 
I find it may due to the conflict between boost 1.62 (updated) and boost 1.54 (original) which I am still debugging. Sorry if I have mislead you.

"
/usr/bin/ld: warning: libboost_thread.so.1.54.0, needed by /opt/ros/indigo/lib/libroscpp.so, may conflict with libboost_thread.so.1.62.0
/usr/bin/ld: warning: libboost_filesystem.so.1.54.0, needed by /opt/ros/indigo/lib/libroscpp.so, may conflict with libboost_filesystem.so.1.62.0
/usr/bin/ld: warning: libboost_system.so.1.62.0, needed by /usr/local/lib/libboost_filesystem.so, may conflict with libboost_system.so.1.54.0
/usr/bin/ld: warning: libboost_iostreams.so.1.54.0, needed by /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../../lib/libpcl_io.so, may conflict with libboost_iostreams.so.1.62.0
/usr/bin/ld: warning: libboost_chrono.so.1.54.0, needed by /usr/lib/gcc/x86_64-linux-gnu/4.8/../../../../lib/libpcl_io.so, may conflict with libboost_chrono.so.1.62.0
"

I install a new ROS indigo in a virtual machine and autoware runs well.

closed 2017-04-06 by amc-nu