autowarefoundation/autoware #554: "ndt_matching process has died" during the quick start.

WeirdAI, 2017-01-06:
I have two laptop to run autoware quick start. They have the same settings. One works normally, but the other always comes to a crash, with the terminal saying 

> ndt_matching: /build/pcl-1.7-pY82A6/pcl-1.7-1.7.1/kdtree/include/pcl/kdtree/impl/kdtree_flann.hpp:172: int pcl::KdTreeFLANN<PointT, Dist>::radiusSearch(const PointT&, double, std::vector<int>&, std::vector<float>&, unsigned int) const [with PointT = pcl::PointXYZ; Dist = flann::L2_Simple<float>]: Assertion `point_representation_->isValid (point) && "Invalid (NaN, Inf) point coordinates given to radiusSearch!"' failed.
> [ndt_matching-6] process has died [pid 19718, exit code -6, cmd /home/hirainbjvr/Autoware/ros/devel/lib/ndt_localizer/ndt_matching __name:=ndt_matching __log:=/home/hirainbjvr/.ros/log/b129ad58-d322-11e6-ab03-484520fe8341/ndt_matching-6.log].
> log file: /home/hirainbjvr/.ros/log/b129ad58-d322-11e6-ab03-484520fe8341/ndt_matching-6*.log

[image]
[image]
[image]
[image]

The terminal, rqt and rviz screenshots are like that.
This problem has stop me from going on the further Autoware functions for more three monthes. Please somebody help me. 
# Please!

WeirdAI, 2017-01-06:
@kitsukawa

kitsukawa, 2017-01-07:
I have never seen the error. It looks like PCL 1.7.1 is installed. Can you upgrade PCL to 1.7.2?
How to upgrade PCL is written in the following page.
https://github.com/CPFL/Autoware/wiki/FAQ(EN)

WeirdAI, 2017-01-10:
Thank you for your help. I'll try. Thanks a lot.

muralijp, 2018-01-31:
@kitsukawa, I have exactly same error. I did install PCL 1.7.2 and ./catkin_make_release. it successfully built and then I changed the PATH like instructed in the link. The problem still remains. Please suggest

muralijp, 2018-01-31:
@WeirdAI. can you update if this solved the problem for you?

yk-fujii, 2018-02-01:
I encountered the same problem.
It became successfully run with re-launch.

I think that quick_start's launch file is not enough initialization for ndt_matching.

yk-fujii, 2018-02-01:
@YamatoAndo Please check it.

YamatoAndo, 2018-02-01:
@muralijp @yk-fujii I can not encounter that problem.
When you  execute "$rostopic echo /filtered_points", is there NaN or Inf in the data?

muralijp, 2018-02-01:
@YamatoAndo 
it says /filtered_points was not published yet. The error description also has
Error: CUDA driver version is insufficient for CUDA runtime version /home/autoware/Autoware/ros/src/computing/perception/localization/lib/fast_pcl/ndt_gpu/src/MatrixDevice.cu 11
should I upgrade CUDA? Pls let me know the process

YamatoAndo, 2018-02-02:
@muralijp 
/points_raw is also not published?

Autoware does not support CUDA 9 yet, so please install CUDA 8.

kitsukawa, 2018-02-20:
@muralijp NDT supports CUDA 9 in #1027 . Please try.

closed 2018-02-20 by kitsukawa

muralijp, 2018-02-20:
Hi Yuki and team,

I need some information on how Lidar data is being preprocessed to generate
points_raw topic. Pls let me know

best regards

JP

kitsukawa, 2018-02-21:
@muralijp How lidar data preprocessed to generate points_raw topic depends on lidar. For example, Velodyne lidars outputs the data in polar coordinates(/velodyne_packets) and then cloud_nodelet converts /velodyne_packets into /points_raw (cartesian cordinates).

tigerk0430, 2018-03-06:
I solved it.

in my case, first of all, I started bag file and pushed a pause button.
then I clicked map & localization button and resumed bag file.

then It worked :-)

liuqi8827, 2019-01-27:
@WeirdAI Hi
I encountered the same problem.

1.My PCL version is 1.7.2
2.I re-build Autoware project many times using ./catkin_make_release.
I started bag file and pushed a pause button. Then I clicked map & localization button and resumed bag file. But the The problem still remains.
I tried many times, the problem still remains.

Did you solve the problem?

best regards.

YamatoAndo, 2019-01-28:
@liuqi8827 Hi
I encountered the same problem last week.
I found the cause of this problem (#1912),
I send a pull request (#1913) last week and merged it into the develop branch (it has not yet been merged into the master branch).

Please try again with the latest develop branch.

gbiggs, 2019-03-06:
Please try the branch referenced above.