autowarefoundation/autoware #558: use Autoware to control a car

Benjaminmar8, 2017-01-09:
Hi, All,
I notice that the in vehilce/zmp directory, there are hv, hv2, mv2, phv. I assume these are different vehicles. If I were to use Autoware to control one of these vehicles, how does the Autoware know which vehicle it is controlling? Thanks.

yukkysaito, 2017-01-29:
@Benjaminmar8 
I think Autoware don't control a vehicle.
Final output topic of Autoware is twist_cmd(geometry_msgs::TwistStamped).
Therefore, control software such as PID is required separately.

closed 2017-01-31 by Benjaminmar8

wsung1, 2018-06-17:
Glad to talk to you, @yukkysaito!
May I give you one quick question regarding this thread?
As you mentioned, the final output of Autoware is twist_cmd.
And twist_cmd is subscribed by vehicle_sender, as shown below.
[image]
My question is how does vehicle_sender connect to vehicle?
In other words, how does vehicle_sender send commands (twist_cmd) to vehicle?
I've walked through vehicle_sender.cpp.
Herein, I could find the codes that are written to subscribe several commands including twist_cmd; however, no codes that work to send the subscribed commands out (to vehicle).
Thanks for your consideration @yukkysaito!

yukkysaito, 2018-06-25:
The node that receives the twist command and controls a vehicle is different for each vehicle. Therefore, it is necessary to prepare a driver node for each vehicle.
Is there a supplement?
@messi49