autowarefoundation/autoware #559: lib way point follower bug

Benjaminmar8, 2017-01-09:
Hi, All,
The way point follower has a bug. The bug can be reproduced when the way point contains repetitious loops. I think this is due to the fact that saved way point has no id. Perhaps when looking up closest way points, we should look ahead a fixed number of way points in front of the vehicle rather than the entire set of way points.

h-ohta, 2017-01-20:
Thank you for your report.
Could you share waypoints file containing repetitious loops?

Benjaminmar8, 2017-01-23:
Here they are. Both waypoint files contains overlapping loops. 
https://drive.google.com/file/d/0B7TRr1J2E914LU5XdmJFNW40OVk/view?usp=sharing
https://drive.google.com/file/d/0B7TRr1J2E914Z2RzTjBjLVpDYnc/view?usp=sharing

h-ohta, 2017-02-02:
Sorry for late reply.

I confirmed waypoint follower was able to run on waypoints containing loops on latest `develop` branch.
In that branch, I fixed getClosestWaypoint() and moved it  into `lane_select` node in lane_planner.
So getClosestWaypoint() in lib waypoint follower has been deprecated.

Could you confirm on `develop` branch?
Thank you for your cooperation.

Benjaminmar8, 2017-02-09:
@h-ohta , I copied the waypoint_follower and waypoint_maker folder in the develop branch into my Autoware src directory. Compilation is ok. But when I use the wf simulator to test the path follower, there was nothing shown in the rviz. In fact, the way points were not even loaded when I run waypoint_loader. What could went run?

h-ohta, 2017-02-09:
@Benjaminmar8 Please also copy `lane_planner ` of `develop` branch.
It has `lane_select` node, including fixes of closest waypoint issue.
After this, Could you record video of your operation?

Benjaminmar8, 2017-04-05:
@h-ohta Hi, sorry for the late reply as I was busy with something else. I just tried what you suggested but the compilation failed. Clearly some of the msg file in the run_time manager package of the devel branch are different from the master branch. 
I glanced through the commits of pass two months and found some are related to the waypoints, but I wasn't sure if these are the fix for the bug mentioned here... Could you advise on that? Thank you very much.

closed 2020-05-29 by gbiggs