autowarefoundation/autoware #560: RVIZ crashes on loading default.rviz

o8r, 2017-01-09:
RVIZ suddenly crashes on loading ros/src/.config/rviz/default.rviz. in step 3.2 of the QuickStart demo. The messages left were:
```
[ INFO] [1483705468.774296510]: rviz version 1.11.15
[ INFO] [1483705468.774336451]: compiled against Qt version 4.8.6
[ INFO] [1483705468.774345900]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1483705469.009310064]: Stereo is NOT SUPPORTED
[ INFO] [1483705469.009366250]: OpenGl version: 3 (GLSL 1.3).
['sh', '-c', '$(rospack find runtime_manager)/../../../.config/rviz/cmd.sh']
pid=15636
sched policy=OTHER prio=0
pid=15637
sched policy=OTHER prio=0

(rviz:15644): GLib-CRITICAL **: Source ID 521 was not found when attempting to remove it

(rviz:15644): GLib-CRITICAL **: Source ID 539 was not found when attempting to remove it

(rviz:15644): GLib-CRITICAL **: Source ID 556 was not found when attempting to remove it
```

Please let me know how to solve the problem.

## Steps to repro the crash
I cleanly installed Ubuntu 14.04 on an amd64 machine and set up Autoware by following steps. The machine has a Geforce GT 420 graphic card but I didn't install CUDA.

Firstly I installed OpenCV 2.4.13 from source and also installed GLEW and Qt5 by:
```
sudo apt-get update
sudo apt-get install libglew-dev qt5-default
```
Installing ROS and dependencies for Autoware,
```
sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu trusty main" > /etc/apt/sources.list.d/ros-latest.list'
wget http://packages.ros.org/ros.key -O - | sudo apt-key add -
sudo apt-get update
sudo apt-get install ros-indigo-desktop-full
sudo rosdep init
rosdep update

sudo apt-get install ros-indigo-nmea-msgs ros-indigo-nmea-navsat-driver ros-indigo-sound-play ros-indigo-jsk-visualization
sudo apt-get install libnlopt-dev freeglut3-dev qtbase5-dev libqt5opengl5-dev libssh2-1-dev libarmadillo-dev libpcap-dev gksu libgl1-mesa-dev
```
and I successfully built Autoware.
Then, I executed steps 2.1--3.2 following QuickStart.

On the other hand, I also set up the same environment on a Hyper-V virtual machine; RVIZ successfully opened the config file in this environment.

o8r, 2017-01-13:
Sorry. I used not Geforce GT 420 but Geforce 740.

yukkysaito, 2017-01-29:
I know the same crash.
There are cases to solve by playing rosbag a little.

I think default.rviz has the problem.

o8r, 2017-03-03:
Thank you for your reply.

I reinstalled Ubuntu14.04 and Autoware and ran RVIZ after playing rosbag to 5%. However, RVIZ crashed again.

Log messages obtained were:

```
['sh', '-c', '$(rospack find runtime_manager)/../../../.config/rviz/cmd.sh']
pid=17531
sched policy=OTHER prio=0
pid=17532
sched policy=OTHER prio=0
/home/xxx/Autoware/ros/src/util/packages/runtime_manager/../../../.config/rviz/cmd.sh: line 44: 17539 Segmentation fault      rosrun rviz rviz
```

zzkou, 2017-03-09:
I  got the same crash.  When I tried 'rosrun rviz rviz'   in another terminal, specified the 'default.rviz'. The same crash would happen again. Here's the info:
$ rosrun rviz rviz 
[ INFO] [1489087826.288732430]: rviz version 1.11.15
[ INFO] [1489087826.288785230]: compiled against Qt version 4.8.6
[ INFO] [1489087826.288797872]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1489087827.048216091, 1427158098.673704139]: Stereo is NOT SUPPORTED
[ INFO] [1489087827.048296199, 1427158098.673704139]: OpenGl version: 4.5 (GLSL 4.5).
Segmentation fault (core dumped)

I haven't figured  out a fix, am doing search on 'rosrun rviz rviz, segmentation fault'

anand042, 2017-04-06:
Confirmed. Autoware comes up successfully, RViz launches and then immediately crashes upon loading default.rviz.

Help would be appreciated!

applekyo, 2017-04-24:
I have met this problem too. And I got a resolution like this:
Change the last two lines of $path/Autoware/ros/src/.config/rviz/default.rviz, which like:
X: 1895
Y: 44

to smaller numbers:
X: 100
Y: 100

save the file. And load it from rviz again. Wish this could help you.

nicolasj92, 2017-05-07:
I have the same problem

aztinki, 2017-07-18:
Did you guys find a solution to this problem? I have the same problem!

closed 2018-04-05 by yk-fujii

123312423, 2018-10-14:
> I got the same crash. When I tried 'rosrun rviz rviz' in another terminal, specified the 'default.rviz'. The same crash would happen again. Here's the info:
> $ rosrun rviz rviz
> [ INFO] [...]

Hi zzkou:

Did you have the solution for this issue ? I have the same issue like yours. Thanks in advance !