autowarefoundation/autoware #565: Question: What tools are you using to create vector maps?

reinzor, 2017-01-16:
Hi,

Thanks for this awesome project! What tools are you using to annotate the grid maps by creating vector maps? I see you save these as .csv file but what are you using to generate these?

Thanks,

Rein

sai5555, 2017-01-26:
I am also interested in this.

yukkysaito, 2017-01-29:
@reinzor @sai5555 
Please wait a moment as I will check

xuhaibin8725, 2017-03-23:
Hi: 
I am also interested in this issue.
I have read and understood the code for resolving the sample data of moriyama vector_map csv files,  but now I have a problem of how to made myself vector map csv files. 

Thanks. 

Haibin.xu

DriftingTimmy, 2017-04-06:
Hi:
I am learning this excellent work and I am also interested in the way they make the map.
Can you share your experience with me about how they transform the information in .csv to the vector map in their program?
Thanks very much. @xuhaibin8725

xuhaibin8725, 2017-04-07:
i write an simple test script using Matlab,  the csv data is from the autoware project and the code demo is as follows. 
i wish it would help you to understand it @DriftingTimmy 

clear all ;clc;
XOFFSET = -1.7e4*0;
YOFFSET = -8.6e4*0;

pointdata = importdata('point.csv');
points = pointdata.data;
lanedata = importdata('lane.csv');
lanes = lanedata.data;
nodedata = importdata('node.csv');
nodes = nodedata.data;
dtlanedata = importdata('dtlane.csv');
dtlanes = dtlanedata.data;
linedata = importdata('line.csv');
lines = linedata.data;
areadata = importdata('area.csv');
areas = areadata.data;
crosswalkdata = importdata('crosswalk.csv');
crosswalks = crosswalkdata.data;

%% lane

figure(100) ;hold on;
bnid=zeros(1,length(lanes(:,1)));
fnid=zeros(1,length(lanes(:,1)));
bpid=zeros(1,length(lanes(:,1)));
fpid=zeros(1,length(lanes(:,1)));
for i = 1 : length(lanes(:,1))
    disp (i/length(lanes(:,1)) );
    bnid(i)=find(nodes(:,1)==lanes(i,5));
    fnid(i)=find(nodes(:,1)==lanes(i,6));
    bpid(i)=find(points(:,1)==nodes(bnid(i),2));
    fpid(i)=find(points(:,1)==nodes(fnid(i),2));
        
end
plot(points(bpid,6)-XOFFSET,points(bpid,5)-YOFFSET,'r*')  ; %%bp
%plot(points(fpid,6)-XOFFSET,points(fpid,5)-YOFFSET,'bo')  ; %%fp

%% dtlane

pid = zeros(1,length(dtlanes(:,1)));
for i = 1 : length(dtlanes(:,1))
    disp (i/length(dtlanes(:,1)));
    pid(i)=find(points(:,1)==dtlanes(i,3));
   
end
% figure(101) ;hold on;
% plot(points(pid,6)-XOFFSET,points(pid,5)-YOFFSET,'g.')  ;

figure(100) ;hold on;
id1 = 1:2636;
id2 = 2637:5278;
id3 = 5279:5532;
id4 = 5533:5545;
id5 = 5546:5580;
id6 = 5581:8302;
id7 = 8303:11026;
id8 = 11027:11046;
id9 = 11047:11065;
id10 = 11066:11110;
id11 = 11111:length(dtlanes(:,1));
% ------
plot(points(pid(id1),6)-XOFFSET,points(pid(id1),5)-YOFFSET,'r.');
plot(points(pid(id2),6)-XOFFSET,points(pid(id2),5)-YOFFSET,'g.');
plot(points(pid(id3),6)-XOFFSET,points(pid(id3),5)-YOFFSET,'m.');
plot(points(pid(id4),6)-XOFFSET,points(pid(id4),5)-YOFFSET,'c.');
plot(points(pid(id5),6)-XOFFSET,points(pid(id5),5)-YOFFSET,'k.');
plot(points(pid(id6),6)-XOFFSET,points(pid(id6),5)-YOFFSET,'y.');
plot(points(pid(id7),6)-XOFFSET,points(pid(id7),5)-YOFFSET,'ro','MarkerSize',3);
plot(points(pid(id8),6)-XOFFSET,points(pid(id8),5)-YOFFSET,'go','MarkerSize',3);
plot(points(pid(id9),6)-XOFFSET,points(pid(id9),5)-YOFFSET,'mo','MarkerSize',3);
plot(points(pid(id10),6)-XOFFSET,points(pid(id10),5)-YOFFSET,'co','MarkerSize',3);
plot(points(pid(id11),6)-XOFFSET,points(pid(id11),5)-YOFFSET,'ko','MarkerSize',3);

%% lines

figure(102) ;hold on;
bpid=zeros(1,length(lines(:,1)));
fpid=zeros(1,length(lines(:,1)));
for i = 1 : length(lines(:,1))
    disp (num2str(i/length(lines(:,1))) );
    bpid(i)=find(points(:,1)==lines(i,2));
    fpid(i)=find(points(:,1)==lines(i,3));
        
    plot([points(bpid(i),6),points(fpid(i),6)],[points(bpid(i),5),points(fpid(i),5)])  ;   
     
end

%% area
figure(103) ;hold on;
slid = zeros(1,length(areas(:,1)));
elid = zeros(1,length(areas(:,1)));
bpid=zeros(1,length(areas(:,1)));
fpid=zeros(1,length(areas(:,1)));
for i = 1 : length(areas(:,1))
    disp (num2str(i/length(areas(:,1))) );
    slid(i)=find(lines(:,1)==areas(i,2));
    elid(i)=find(lines(:,1)==areas(i,3));
    for j =slid(i):elid(i)
    bpid(i)=find(points(:,1)==lines(j,2));
    fpid(i)=find(points(:,1)==lines(j,3));
        
    plot([points(bpid(i),6),points(fpid(i),6)],[points(bpid(i),5),points(fpid(i),5)])  ;   
    end
end

%% crosswalk
figure(104) ;hold on;
aid = zeros(1,length(crosswalks(:,1)));
slid = zeros(1,length(crosswalks(:,1)));
elid = zeros(1,length(crosswalks(:,1)));
bpid = zeros(1,length(crosswalks(:,1)));
fpid = zeros(1,length(crosswalks(:,1)));
for i = 1 : length(crosswalks(:,1))
    disp (num2str(i/length(crosswalks(:,1))) );
    aid(i) = find(areas(:,1)==crosswalks(i,2));
    slid(i) = find(lines(:,1)==areas(aid(i),2));
    elid(i) = find(lines(:,1)==areas(aid(i),3));
    for j = slid(i):elid(i)
    bpid(i) = find(points(:,1)==lines(j,2));
    fpid(i) = find(points(:,1)==lines(j,3));
        
    plot([points(bpid(i),6),points(fpid(i),6)],[points(bpid(i),5),points(fpid(i),5)])  ;   
    end
end

DriftingTimmy, 2017-04-07:
Thanks very much.It will help me a lot.
But I am more interested in the way that how the Autoware works by these .csv and their components' meanings.Can you show me these ways?
If you can leave your email,that will be easier for me to contact you.
And I think you are Chinese right?
Or you can contact me through my email adr. 605383972@qq.com.
@xuhaibin8725

DriftingTimmy, 2017-04-10:
After spending some time learning the code in MATLAB, I can understand a part of the data structure. But I still have some questions like what the meaning of the data in 'node.csv' and some other questions, can you teach me how to connect all the information together into a vector map?
@xuhaibin8725   Thanks very much!

westeast, 2017-04-11:
I am also interested in this. Are there any official docs? @xuhaibin8725 . so that  I can generate my own vector_map .

DriftingTimmy, 2017-04-11:
Have you understood the exact meaning of all factors in the csv file?

bobjohnson2040, 2017-05-01:
I am also very interested in learning more about the vector maps! How do you create the CSV?

There are many people interested in this, so please keep the conversation here instead of on private email @DriftingTimmy

DriftingTimmy, 2017-05-02:
I got this, but he didn't reply me anyway. I just figure out the relationship between the different .csv file.

bobjohnson2040, 2017-06-02:
@DriftingTimmy Can you please share anything that you were able to figure out?

arashroomi, 2017-11-27:
Does anyone knows what does each value in the CSV files mean? Specially in the dtlane.csv and lane.csv
Imade some guesses fro example Slope or RefVel, but do you guys have also ideas about the csv file?

ahuer2435, 2018-01-04:
Autoware seems to provide a tools for product vector map from PCD files, as : https://maptools.tier4.jp/vector_mapper_description/ .
But I could not use it yet, it prompts " Now Creating... Please wait for creating your environment."

kfunaoka, 2018-07-13:
VectorMapper is ready to use!
https://maptools.tier4.jp/vector_mapper_description/

closed 2018-07-13 by kfunaoka