autowarefoundation/autoware #572: obstacle detection and car control

Benjaminmar8, 2017-01-23:
Hi, All,
How does Autoware detects an obstacle (car, pedestrian, buildings, etc) and slows down or stops the car if it is necessary? Is there something similar to costmap as the ROS navigation stack? Thanks a lot for answering my question.

mslavescu, 2017-01-23:
I'm interested in this also, and how easy is to plug new algorithms in Autoware.
How can I use the 3D depth map from Autoware mono camera (not LIDAR) ORB_SLAM output to detect cars, pedestrians, and other objects?

yukkysaito, 2017-01-29:
@Benjaminmar8 
Autoware has two areas(slows down, stops).
If an obstacle exist these areas, it controls to slow down or stop.

Today I add the costmap to Autoware develop branch.
However, cooperation with the local planning has not been completed yet.
@mslavescu 
I think ORB SLAM can output pointcloud2 of feature points, this pointcloud2 topic can be substituted for points_raw published from Lidar

yukkysaito, 2017-01-29:
https://www.youtube.com/watch?v=tW7S74EUH3Y

Benjaminmar8, 2017-01-31:
@yukkysaito 
Thanks for the reply. Let's say a recorded path is given and the car is supposed to follow this path, will the car slow down or stop without using the costmap?

mslavescu, 2017-01-31:
Thanks @yukkysaito!

Do you have a dataset/instructions to reproduce the behavior in the video?

Have you looked into this method?
Multi-Level Mapping: Real-time Dense Monocular SLAM
https://m.youtube.com/watch?v=qk2ViPVxmq0

How it compares with ORB_SLAM2 (Monocular) used in Autoware?

yukkysaito, 2017-02-02:
@Benjaminmar8 In the current implementation, it will. However, we are also developing a method to use a cost map.
like this (https://www.youtube.com/watch?v=TA0_0Zb7cjE). this video is not Autoware.

yukkysaito, 2017-02-02:
@mslavescu I am interested in this video.
But it cannot use to detect objects because it seems there is a big delay.
I think it can use create a 3D map.

@sujiwo develop ORB SLAM in Autoware. Therefore he may be able to give good advice.
What do you think?

mslavescu, 2017-02-02:
Thanks @yukkysaito!
They mentioned realtime 3D reconstruction at 30Hz without GPU, would not be that enough to imput to detectors (not sure how dense is the point cloud, if it is ebough or not)?

@sujiwo would be great if you can give me some advice how to plug some monocular videos in Autoware ORB_SLAM2 implementstion, and if is possible to do 3D reconstruction in Autoware, and maybe plug the generated point clouds into the LIDAR based detectors implemented in Autoware.

I would like to integrate this functionality in the simulator shown here, which I also plan to use in raceai.net to attract gamers in testing OSSDC software:
https://mobile.twitter.com/GTARobotics/status/827250102989422592

yukkysaito, 2017-02-02:
@mslavescu for your reference. The following video is ORB SLAM developed by @sujiwo 
https://www.youtube.com/watch?v=2x_KvXACAgc

mslavescu, 2017-02-02:
@yukkysaito thanks! That video looks amazing!
@sujiwo would you be able to share some instructions on how to do what your did in the video with your dataset, but also on my own videos, including those from PS3/PS4 driving simulator games?

closed 2019-03-01 by gbiggs