autowarefoundation/autoware #578: Could NOT find GLEW (missing: GLEW_INCLUDE_DIR GLEW_LIBRARY)

xonez, 2017-01-31:
Hi,

I am trying to running the autoware, after     ./catkin_make_release
but comes to this error, could someone give me some ideas?

best

bbidault2, 2017-01-31:
Hi @xonez, this is a very common error when installing autoware, you need to install libglew-dev,
`sudo apt-get install libglew-dev`

xonez, 2017-01-31:
Hi, thanks so much, are there instruction of how to quick start the autoware?
After I install the autoware, ./run, the quick start need me to type the cmd_parm,
where is the cmd_parm?

bbidault2, 2017-02-03:
I am not sure you are following the right doc, you should check out this document -> https://github.com/CPFL/Autoware/blob/master/docs/en/autoware-quickstart.pdf

yk-fujii, 2017-05-29:
we added it to the readme and we will add it to the document in the future.
Thanks

closed 2017-05-29 by yk-fujii