autowarefoundation/autoware #612: No gui after install the docker version.

huyong1109, 2017-03-13:
I followed the installation guides in the docker subdir. After successful installation, how could I run Autoware in docker? I get this:
```
autoware@8c10d4c3f390:~/Autoware/ros$ ./run
Process Manager

(gksudo:133): Gtk-WARNING **: cannot open display: localhost:11.0

** (gnome-terminal:135): WARNING **: Could not open X display

** (gnome-terminal:136): WARNING **: Could not open X display
error: XDG_RUNTIME_DIR not set in the environment.
Failed to parse arguments: Cannot open display: 
error: XDG_RUNTIME_DIR not set in the environment.
Failed to parse arguments: Cannot open display: 
```

I guess that maybe a X-window version of ubuntu required in Docker?  Thanks!

BTW.  some dependencies are missed in the Dockerfile. 

```
RUN sudo apt-get install -y ros-indigo-grid-map ros-indigo-controller-manager \
          ros-indigo-ros-control ros-indigo-ros-controllers ros-indigo-gazebo-ros-control \
         ros-indigo-sicktoolbox ros-indigo-sicktoolbox-wrapper ros-indigo-joystick-drivers ros-indigo-novatel-span-driver \
          libnlopt-dev freeglut3-dev qtbase5-dev libqt5opengl5-dev libssh2-1-dev libarmadillo-dev libpcap-dev gksu libgl1-mesa-dev libglew-dev
```

ClementLeBihan, 2017-05-30:
Same error on my side :/

malichao, 2017-06-02:
Same issue here.
```
** (gnome-terminal:1): WARNING **: Could not open X display
No protocol specified
error: XDG_RUNTIME_DIR not set in the environment.
Failed to parse arguments: Cannot open display: 
```
Apart from that, I need to manually install a few missing dependencies:
```
-FROM nvidia/cuda:8.0-ubuntu14.04
+FROM nvidia/cuda:8.0-devel-ubuntu14.04

# Install some required ROS packages
RUN apt-get install -y \
			ros-indigo-grid-map \
			ros-indigo-velocity-controllers \
			ros-indigo-gps-common \
			libglew-dev
```

uriv, 2017-06-14:
Here's how I fixed it. In the Dockerfile, find these lines:
```
        # Replace 1000 with your user/group id
        usermod  --uid 1000 $USERNAME && \
        groupmod --gid 1000 $USERNAME
```
The Linux user id of user autoware ($USERNAME) needs to match in the host OS and in Docker. Here, the number 1000 is set by default. The user id can be found (outside of Docker) with the command: `id -u autoware`. For me, it was 1002. So, I had to change 1000 to 1002. 

To automate the change, update the block in Dockerfile with the following and rebuild with `./build.sh`. This may take a lot of time because all following blocks would need to be rebuilt. See a faster way below.
```
RUN USERID=$(id -u $USERNAME) && \
	useradd -m $USERNAME && \
        echo "$USERNAME:$USERNAME" | chpasswd && \
        usermod --shell /bin/bash $USERNAME && \
        usermod -aG sudo $USERNAME && \
        echo "$USERNAME ALL=(ALL) NOPASSWD:ALL" >> /etc/sudoers.d/$USERNAME && \
        chmod 0440 /etc/sudoers.d/$USERNAME && \
        # Replace 1000 with your user/group id
        usermod  --uid $USERID $USERNAME && \
        groupmod --gid $USERID $USERNAME
```
A fix with shorter build time would be to put: 
```
RUN 	USERID=$(id -u $USERNAME) && \
	sudo usermod  --uid 1002 $USERNAME && \
	sudo groupmod --gid 1002 $USERNAME
```
**before (!)** `RUN gconftool ...`
then rebuild with `./build.sh`

huckl3b3rry87, 2017-11-28:
Using the master branch of Autoware, I run into similar issues:
```
febbo@febbo-HP-ZBook-17-G2:~/Autoware/docker$ ./run.sh

** (gnome-terminal:1): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-oTmiGTHcd2: Connection refused

(gnome-terminal:1): GLib-GIO-CRITICAL **: g_settings_get: the format string may not contain '&' (key 'monospace-font-name' from schema 'org.gnome.desktop.interface'). This call will probably stop working with a future version of glib.
```

nyoshimura, 2018-05-23:
This should be written in readme...

nyoshimura, 2018-05-24:
For me, solved by run `xhost +` before `./run`

wmpang, 2018-09-06:
I tried invoking xhost + first, but seems I got following error message:

access control disabled, clients can connect from any host

chenz16, 2019-01-02:
tested and passed:  run "xhost +" (NON-docker env) and the system provided the message "access control disabled, clients can connect from any host". Leave as it is. Then go to docker env and "sudo ./run"

filiperinaldi, 2019-02-05:
@huyong1109, Are you still having this issue?
@wmpang, that is not an error message, that is telling you *access control is disabled* so your container should now be able to connect to your X server. Are you still facing this issue?

hahaxun, 2019-07-29:
> For me, solved by run `xhost +` at the outside of docker, then activate docker & `./run`

according -e DISPLAY=$DISPLAY -v /tmp/.X11-unix/:/tmp/.X11-unix:ro and  "xhost +" at the outside of docker, everything is OK.

closed 2019-08-04 by gbiggs