autowarefoundation/autoware #617: trajectory planning module

yyuuliang, 2017-03-22:
Thanks for making this project open source. I have learned a lot from the code.
Now I am studying the lattice trajectory generating code, is there any published paper from the code? or is this approach based on some other's work? Can I have the paper title?
Thanks :)

yyuuliang, 2017-03-23:
got answered, thanks!

closed 2017-03-23 by yyuuliang

gulshan216, 2018-06-13:
I am studying Lattice based trajectory generation as well. Could you please let me know the reference paper used for implementation in Autoware. @yyuuliang Would appreciate your help here.

lijiaming5550, 2022-08-03:
I am studying Lattice based trajectory generation as well.  When I read the "lattice_planner" algorithm, I am not very clear about the origin of some formulas involved in "libtraj_gen.cpp". According notes, for example: Thomas howard's discrimination Page 38, these equations can be found in McNaughton, page 76 and compute next command from Howard page 39,......I think these formulas may come from a book or material. Could you please let me know the reference paper used for implementation in Autoware. @yyuuliang Would appreciate your help here