autowarefoundation/autoware #625: why the builded map data  used the ndt_mapping data is so big?

thomas3016, 2017-04-09:
Deal all,
When I build the map using the ndt_mapping according to the video, I found that the builded map will be over 1G using the sample data. Wether any place I went wrong? Any help will be appreciated!

kitsukawa, 2017-04-09:
You can reduce the file size using voxel grid filter afterwards. It is in the Map tab.

thomas3016, 2017-04-09:
@kitsukawa thank you for your reply, I have try to reduce the file size using PCD Filter. The size of my pcd file 1.3G and I set the leaf size as 0.2,0.3,0.4,0.45,0.48,respectively, after filter, I got the same size of pcd file(635.7M). When I set the leaf size as 0.5, I got a pcd file  (10.8M). So the first got pcd file is too thick and the second file I got is too sparse. I really have no idea, can you give me some suggestion

DriftingTimmy, 2017-05-11:
Hi, can you tell me how to run the ndt_mapping with my own data? And is there any guidance I can refer to? Thanks a lot.

kitsukawa, 2017-05-11:
See this video.
https://www.youtube.com/watch?v=HlQ0ohxvlgA

DriftingTimmy, 2017-05-11:
Thanks a lot, I m trying to do some experiments, and I really appreciate the excellent work you and your team achieved! @kitsukawa  If I have other problems, I will be back~ hhh

DriftingTimmy, 2017-05-12:
Oh, and there is one more question, can I use my '.pcap' file to run autoware mapping and then save the result? @kitsukawa

kitsukawa, 2017-05-12:
Yes. First, convert pcap to rosbag.
http://answers.ros.org/question/213080/convert-raw-velodyne-vlp16-pcap-to-bagfile/
The rosbag includes the topic, velodyne_packets, so you have to convert the topic from /velodyne_packets to /points_raw using VLP-16 driver in Sensing tab.

DriftingTimmy, 2017-05-15:
That's really helpful, but I just want to use ndt_mapping or icp_mapping with my own data only, instead of running Autoware everytime, but everytime I try to run ndt_mapping isolated, there's always an error of : 
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "runtime_manager"
  with any of the following names:

    runtime_managerConfig.cmake
    runtime_manager-config.cmake

  Add the installation prefix of "runtime_manager" to CMAKE_PREFIX_PATH or
  set "runtime_manager_DIR" to a directory containing one of the above files.
  If "runtime_manager" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  ndt_localizer-master/CMakeLists.txt:4 (find_package)

If I annotate the cmake file, it can not run successfully.

kitsukawa, 2017-05-16:
ndt_localizer package depends on runtime_manager package in relation to message generation. So can you build the all packages in Autoare using ./catkin_make_release in Autoware/ros directory?

DriftingTimmy, 2017-05-17:
Yes I can, after a few correction, I can run the ndt_localizer package, but I still need some modification of my code to satisfy my dataset. Thank you a lot @kitsukawa

closed 2017-05-17 by thomas3016