autowarefoundation/autoware #627: Where can I find data format of Vector Map

so1987, 2017-04-10:
Thank you for developers everyday.

I want to a little customize vector map of moriyama-sample.
I checked csv data; point.csv, area.csv, line.csv, lane.csv, gutter.csv, crosswalk.csv,,, etc.

But their data format is really not understandable.
e.g., point.csv has columns named PID, Bx, Ly, B, H, L, ReF, MCODE1, MCODE2, MCODE3.
What I can guess is only that PID stands for Point ID.

line.csv has LID, BPID, FPID, BLID, FLID, similarly not understandable. Others are the same.

I really appreciate if anyone help me where or how I can understand them.

Thank you in advance.

xuhaibin8725, 2017-04-10:
line.csv
Lid: line id
Bpid: the beginning point's I'd
For the line
Food: the finishing point's I'd for the line
Blid: the id of the line before this line
Flid: the id of the line after this line

发自网易邮箱大师
On 04/10/2017 20:08, so1987 wrote:

Thank you for developers everyday.

I want to a little customize vector map of moriyama-sample.
I checked csv data; point.csv, area.csv, line.csv, lane.csv, gutter.csv, crosswalk.csv,,, etc.

But their data format is really not understandable.
e.g., point.csv has columns named PID, Bx, Ly, B, H, L, ReF, MCODE1, MCODE2, MCODE3.
What I can guess is only that PID stands for Point ID.

line.csv has LID, BPID, FPID, BLID, FLID, similarly not understandable. Others are the same.

I really appreciate if anyone help me where or how I can understand them.

Thank you in advance.

so1987, 2017-04-10:
Great, thank you for the answer! That makes sense!
But what we can do is only guessing?

Anyway, I appreciate you @xuhaibin8725.

westeast, 2017-04-11:
Are there any official documents about the vector_map?

prarobo, 2017-09-12:
I am having the same problem with understanding vector maps. Can someone give an explanation for the fields in dtlane.csv?

The fields in the file are 
DID | Dist | PID | Dir | Apara | r | slope | cant | LW | RW

Thanks.

nangong, 2017-09-26:
@prarobo Have u already found out the meaning?

prarobo, 2017-09-26:
@histfuture No I do not know the meaning of those parameters.

tobyhijzen, 2018-05-08:
I looked into the data format a bit. So far I have figured out that point data contains:

- PID: Point ID
- B: Latitude
- L: Longitude
- H: Height
- Bx: x coordinate where the map was created in. 
- Ly: y coordinate where the map was created in.
- ReF: ???Short for reference, but to what???
- MCODE[1,2,3]: ?

For the moriyama sample dataset the x and y coordinates corresponds to the "Japanese rectangular coordinate system".

Interestingly the `MCODE1` in the moriyama sample dataset is either 5236 or 5237 and this corresponds to the longitude being smaller than 137 (->5236) or bigger than 137 (->5237). The `MCODE` values are still a mystery to me.

jraclarke, 2018-05-08:
@tobyhijzen   I'm not sure about the MCODEs, but I know that the ReF refers to the Japanese Rectangular Plane number - so for example 7 refers to "JGD2011 / Japan Plane Rectangular CS VII" (https://epsg.io/6675).  In total there are 19 planes that cover Japan.

gbiggs, 2019-03-01:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions on the [Autoware Discourse category](https://discourse.ros.org/c/autoware) following our support guidelines: https://github.com/CPFL/Autoware/wiki/Support-guidelines. Please pay particular attention to the information we ask you to provide.

closed 2019-03-01 by gbiggs

zlg9folira, 2021-06-07:
Any progress on this? Same issue here, what exactly are `MODE1` `MODE2` and `MODE3`. I am trying to create my own map  for Autoware 1.12. I first created a map using lanelet and converted it to 1.12 CSV version. However, all these three columns are zero and I need to know what they are to modify.

JWhitleyWork, 2021-06-07:
@zlg9folira Please do not revive old threads. To get support, your first avenue should be a post on [ROS Answers](https://answers.ros.org) with the tag `autoware`.