autowarefoundation/autoware #629: An error was occured when build mapping using ndt_mapping

thomas3016, 2017-04-12:
Deal all, when I build mapping using ndt_mapping with given sample data, an error that "a message of over a gigabyte was predicted in tcpros. that seems highly unlikely, so I'll assume protocol synchronization is lost" was occrured. who can tell me why, any help will be appreciated!

thomas3016, 2017-04-13:
@Benjaminmar8 @kitsukawa  can you help me to answer this question? I shall be very grateful.

kitsukawa, 2018-02-20:
@thomas3016 Due to ROS specification, if messages over 1GB is published, the error you mentioned occurs. ndt_mapping node publishes all the point cloud previously built, so if the map becomes larger up to a certain size, the message can not be published. If you want to create wide area map, you can use approximate_ndt_mapping instead of ndt_mapping. The node loads and publishes a part of the map, so the error above does not occur.

closed 2018-02-20 by kitsukawa