autowarefoundation/autoware #638: Error when playing demo

nicolasj92, 2017-04-30:
The demo plays but the console shows the following output:

```
[ndt_matching-6] process has died [pid 4068, exit code -6, cmd /home/nj/autoware/Autoware/ros/devel/lib/ndt_localizer/ndt_matching __name:=ndt_matching __log:=/home/nj/.ros/log/bec90b44-2dcb-11e7-9f2f-001c428ebb76/ndt_matching-6.log].
log file: /home/nj/.ros/log/bec90b44-2dcb-11e7-9f2f-001c428ebb76/ndt_matching-6*.log
QVariantMap DBusMenuExporterDBus::getProperties(int, const QStringList&) const: Condition failed: action
QVariantMap DBusMenuExporterDBus::getProperties(int, const QStringList&) const: Condition failed: action

```

kitsukawa, 2017-05-04:
Try to compile using ./catkin_make_release and execute again?

nicolasj92, 2017-05-04:
The Error is still the same (ntd_matching dies):
`Invalid (NaN, Inf) point coordinates given to radiusSearch!"' failed.
`

nicolasj92, 2017-05-05:
New installation of ubuntu and ros solved it somehow

closed 2017-05-05 by nicolasj92