autowarefoundation/autoware #652: Docerfile

andoh104, 2017-05-20:
Dear all,  this is my first Issues.

I installed Autoware in Ubuntu 16.04 docker. Then, because of the lack of packages in Dockerfile, it was impossible to build Autoware.
The missing packages are shown below.

`RUN apt - get install - y libnlopt - dev freeglut 3 - dev qtbase 5 - dev \`
`     Ros-indigo-sicktoolbox ros-indigo-sicktoolbox-wrapper`

The following modifications were also required for Dockerfile.
Https://github.com/CPFL/Autoware/pull/580/files

andoh104, 2017-05-23:
There were some mistake in the previous comment, so I  edited it.

yk-fujii, 2017-05-28:
Could you make a Pull Request?
Thanks.

andoh104, 2017-05-28:
I do not know which repository I should push my changes.
Could you tell me how to Pull Request?

yk-fujii, 2017-05-29:
Autoware project adopt the following branch model:
https://github.com/CPFL/Autoware/blob/develop/docs/jp/branching_model.md

Please make a PR to develop branch after creating your fork repository and committing modifications.
refer to the document below:
https://help.github.com/articles/creating-a-pull-request/
https://help.github.com/articles/about-pull-requests/
http://kik.xii.jp/archives/179  (Japanese document)

thanks

andoh104, 2017-05-29:
I made a fork and fix branch and Pull Request.
https://github.com/andoh104/Autoware/tree/fixDocker

yk-fujii, 2017-05-30:
Thank you for your contribution!

but, I saw the PR which was creating on your repositories.....
Could you make PR on the our CPFL/Autoware repositories?

andoh104, 2017-05-30:
I did Pull request to develop branch.  
I'd like to commit only Dockerfile,  but some other files were requested.
Please merge properly.

yk-fujii, 2017-05-30:
Thanks! I had a merge it!

closed 2017-05-30 by yk-fujii