autowarefoundation/autoware #659: Runtime Manager

naoto3, 2017-05-26:
Autowareビルド後、Runtime Managerを起動しようとしても、パスワードが入力が求められ、端末が一つ表示されるだけで何も出力されません。エラーは出ていません。

manato, 2017-05-27:
@naoto3 ,
Autoware内で御使用のの環境におけるタスクの優先度を操作するためにパスワードが求められます．
御使用の環境でのパスワード（ユーザパスワード．`sudo`権限を持っている事が必要）を入力してみていただけますか？

naoto3, 2017-05-27:
返信有難う御座います。
パスワードを入力してみましたが、何も出力されませんでした。

manato, 2017-05-27:
2点伺いたい事がございます：
* 御使用のユーザには`sudo`権限が付与されていますでしょうか？
* パスワード入力後に，一番はじめに立ち上がる端末には何か表示されていませんでしょうか？

naoto3, 2017-05-28:
返信が遅くなって申し訳ありません。

>・御使用のユーザにはsudo権限が付与されていますでしょうか？
確かにされています

>・パスワード入力後に，一番はじめに立ち上がる端末には何か表示されていませんでしょうか？
端末には以下のような表示がされています。
---------------------------端末---------------------------------
... logging to /home/ユーザー名/.ros/log/5714a5ca-4212-11e7-8107-4ccc6a66c3d9/roslaunch-ユーザー名-Z170-S01-6817.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ユーザー名-Z170-S01:38848/
ros_comm version 1.11.21

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES

auto-starting new master
process[master]: started with pid [6841]
ROS_MASTER_URI=http://ユーザー名-Z170-S01:11311/

setting /run_id to 5714a5ca-4212-11e7-8107-4ccc6a66c3d9
process[rosout-1]: started with pid [6854]
started core service [/rosout]
-----------------------------------------------------------------------

yk-fujii, 2017-05-28:
rosmasterが立ち上がっている（端末で上記ログが出ている）状態で
```
source devel/setup.bash  
rosrun runtime_manager runtime_manager_dialog.py
```
を実行するとエラー等発生しますか？

manato, 2017-05-28:
表示内容の御送付ありがとうございます．
それでは以下はいかがでしょうか？
* プログラムの起動はどのように行われていますか？  
  * 基本的には`Autoware/ros/run`を実行して下さい
* 上記シェルで実行している場合：
  * 上記シェル中では少なくとも2つの端末が開くようになっていますが，ひとつしか立ち上がらない（様に見える）と言うことは，片方の端末でエラーが発生し，そのまま閉じている可能性がございます．
  * `Autoware/ros/run`の最終行を以下に変更して，表示内容を御確認いただけますでしょうか？  
    最後の部分を`"bash -c 'source ./devel/setup.bash; rosrun runtime_manager runtime_manager_dialog.py'"` → `"bash -c 'source ./devel/setup.bash; rosrun runtime_manager runtime_manager_dialog.py; sleep 1000;'"`

manato, 2017-05-28:
@yk-fujii 
コメントありがとうございます．
恐らくrosmasterは立ち上がっていて，runtime-managerの起動に失敗→エラーが発生した瞬間に2つ目の端末が落ちている  
的な状況ではないかと…

naoto3, 2017-05-29:
@manato 
返信有難うございます。

runの最終行を変更したところ
ImportError: No module named wx
AttributeError: 'module' object has no attribute 'NUM_CPUS'
のエラーが発生していることが分かり、エラーを修正したところ無事runtime-managerが起動しました。

有難うございました。

closed 2017-05-29 by naoto3