autowarefoundation/autoware #660: Could not find a package configuration file provided by "grid_map_ros"

zdx3578, 2017-05-28:
dockerfile add : 

RUN sudo apt-get -y install ros-indigo-grid-map

andoh104, 2017-06-01:
This issue has been fixed in the following.
https://github.com/CPFL/Autoware/pull/661

But it has not been merged in the master branch yet. I don't know when it will be merged.

andoh104, 2017-06-02:
It has already been merged in the master branch.
https://github.com/CPFL/Autoware/commit/e05cf7b7d6c065fa3f4d7abe92ba0f3f9a4f753c#diff-ebacf6f6ae4ee68078bb16454b23247d

Thanks,

zdx3578, 2017-06-04:
Could not find the required component 'gps_common'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "gps_common" with
  any of the following names:

    gps_commonConfig.cmake
    gps_common-config.cmake

  Add the installation prefix of "gps_common" to CMAKE_PREFIX_PATH or set
  "gps_common_DIR" to a directory containing one of the above files.  If
  "gps_common" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  sensing/drivers/imu/packages/xsens/src/xsens_driver/CMakeLists.txt:7 (find_package)

add below from readme:

sudo apt-get install ros-indigo-desktop-full ros-indigo-nmea-msgs ros-indigo-nmea-navsat-driver ros-indigo-sound-play ros-indigo-jsk-visualization ros-indigo-grid-map ros-indigo-gps-common
 sudo apt-get install ros-indigo-controller-manager ros-indigo-ros-control ros-indigo-ros-controllers ros-indigo-gazebo-ros-control ros-indigo-sicktoolbox ros-indigo-sicktoolbox-wrapper ros-indigo-joystick-drivers ros-indigo-novatel-span-driver
 sudo apt-get install libnlopt-dev freeglut3-dev qtbase5-dev libqt5opengl5-dev libssh2-1-dev libarmadillo-dev libpcap-dev gksu libgl1-mesa-dev libglew-dev

andoh104, 2017-06-05:
I tested the master branch again, then  the error occurred.
I am going to try to fix it.

zdx3578, 2017-06-06:
ubuntu@ip-172-30-0-188:~/github/Autoware/docker$ sudo sh build.sh 
nvidia-docker | 2017/06/06 10:39:23 Error: unsupported Docker version

change docker readme：

use latest nvidia docker:

wget -P /tmp https://github.com/NVIDIA/nvidia-docker/releases/download/v1.0.1/nvidia-docker-1.0.1-1.x86_64.rpm
sudo rpm -i /tmp/nvidia-docker*.rpm && rm /tmp/nvidia-docker*.rpm

andoh104, 2017-06-06:
I have made a pull request .
https://github.com/CPFL/Autoware/pull/680

andoh104, 2017-06-07:
I have made a next pull request to fix README.md.
https://github.com/CPFL/Autoware/pull/683

andoh104, 2017-07-15:
They was merged into master branch.

closed 2018-04-05 by yk-fujii