autowarefoundation/autoware #679: compile error

hriapatil, 2017-06-05:
1. nvcc fatal   : Stray '"' character
CMake Error at rbsspfvehicletracker_generated_rbsspfvehicletracker.cu.o.Release.cmake:222 (message):
  Error generating  /home/apatil/Autoware/ros/build/computing/perception/detection/packages/lidar_tracker/CMakeFiles/rbsspfvehicletracker.dir/nodes/vscan_lidar_track/./rbsspfvehicletracker_generated_rbsspfvehicletracker.cu.o

2. /usr/bin/ld: cannot find -lDummyDevice
/usr/bin/ld: cannot find -lOniFile
/usr/bin/ld: cannot find -lPS1080

kitsukawa, 2017-06-06:
With regard to 2, you can solve the error doing this.
> $ export LIBRARY_PATH=/usr/lib/OpenNI2/Drivers:$LIBRARY_PATH

bholdmanny, 2017-09-26:
@hriapatil Hello, have you solved the error 1. 
nvcc fatal : Stray '"' character
CMake Error at rbsspfvehicletracker_generated_rbsspfvehicletracker.cu.o.Release.cmake:222 (message):
Error generating /home/apatil/Autoware/ros/build/computing/perception/detection/packages/lidar_tracker/CMakeFiles/rbsspfvehicletracker.dir/nodes/vscan_lidar_track/./rbsspfvehicletracker_generated_rbsspfvehicletracker.cu.o