autowarefoundation/autoware #681: Stuck at Mapping

gsrujana, 2017-06-06:
Hi,
I followed the instructions in the manual. I pressed the mapping button. It loads all the pcl files but is stuck on this step for a long time. I never get "OK". 

load /home/psvuser/.autoware/data/map/vector_map/area.csv, areaclasses.size()=924
finished load files
start publish vector map

How do I generate the map? Thanks!

gsrujana, 2017-06-07:
Also, In Rviz it shows status errors in "Points Raw" and "Vehicle Model" as it can't find parameters. I guess that is because the mapping is not done.

closed 2017-06-07 by gsrujana

iulialexandra, 2017-07-25:
Hi, I have the same problem, could you please tell me what you did to load the map?

gsrujana, 2017-07-25:
Hi,

This happens when you don't have the rosbag simulation or any have incoming velodyne/nmss data. Once you start simulation it should work..