autowarefoundation/autoware #682: I can't launch the Runtime Manager

SDwaiwai, 2017-06-07:
When I run the "./run" in the terminal and after enter the supervisor password,there was only one rosnode launched which named roscore,but I can't see the Runtime Manager window,can you help me?My version is ubuntu16.04 and ros kinetic.

manato, 2017-06-08:
@SDwaiwai
Would you modify `Autoware/ros/run` script as following and see what error messages appear?
```
# The final line of "run" script
# Current
${TERMINAL} ${RUNMGR_DISPLAY_OPTION} ${OPTION_WORKING_DIR}=${MY_PATH} ${OPTION_COMMAND}="bash -c 'source ./devel/setup.bash; rosrun runtime_manager runtime_manager_dialog.py'"

# Modify like this
${TERMINAL} ${RUNMGR_DISPLAY_OPTION} ${OPTION_WORKING_DIR}=${MY_PATH} ${OPTION_COMMAND}="bash -c 'source ./devel/setup.bash; rosrun runtime_manager runtime_manager_dialog.py; sleep 1000;'"

# Just add `; sleep 1000;` after boot runtime_manager
# This modification may help us to know what error exactly happened.
```

Best

tenfoldpaper, 2017-06-08:
I had an identical error as the OP. After following manato's advice, I found out that wx tools were not being imported. Running this line fixed the problem:
`sudo apt-get install python-wxtools`

I have a very new install of Ubuntu, so sort of makes sense that I forgot to install couple tools for python here and there.

Hope that helps others.

SDwaiwai, 2017-06-09:
Thanks very much , I fix this problem by 
sudo apt-get install python-wxtools
then I remake it and it works

closed 2017-06-09 by SDwaiwai

Srivatsa-M-Hegde, 2018-10-17:
getting stuck on the exact same issue, error as follows " File "/home/srivatsa/Autoware/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py", line 59, in <module>
    from autoware_msgs.msg import ConfigSsd
ImportError: No module named autoware_msgs.msg"

Has anyone figured out what the issue is ?

hschhh666, 2018-12-07:
@Srivatsa-M-Hegde  I have the same problem as you!!! And I cannot fix it now, do you have any idea?

Srivatsa-M-Hegde, 2018-12-07:
I tried the source install rather than the docker one, that works fine.

hschhh666, 2018-12-07:
@Srivatsa-M-Hegde Thanks! I use the source install, and it works. But when I checkout to version 1.7.0 to run the demo, the problem occurs again.

Srivatsa-M-Hegde, 2018-12-07:
As far as I can remember the source install did not give me this error. It was only with the docker install that I faced this error.

datnguyen82, 2019-05-30:
The run.sh file in Autoware/docker/generic seems to be not working. 
I changed the script to: 

````
#!/bin/sh

XSOCK=/tmp/.X11-unix

XAUTH=/home/$USER/.Xauthority

SHARED_DIR=/home/autoware/.autoware

HOST_DIR=/home/$USER/.autoware

nvidia-docker run \

    -it --rm \
    --volume=$XSOCK:$XSOCK:rw \
    --volume=$XAUTH:$XAUTH:rw \
    --volume=$HOST_DIR:$SHARED_DIR:rw \
    --env="XAUTHORITY=${XAUTH}" \
    --env="DISPLAY=${DISPLAY}" \
    -u autoware \
    --privileged -v /dev/bus/usb:/dev/bus/usb \
    --net=host \
    autoware/autoware:latest-kinetic
```` 
According to this: [link](https://ai4sig.org/2018/07/docker-for-autoware/?fbclid=IwAR2ImpFvYf8D97HIVQ--YhwZpC4bl6VKS5fZ5uGxixVxjRGjxIbf93V-ywU)

Then I ran sudo bash run.sh again in Autoware/docker/generic directory and it led me to the docker terminal.

After that, I ran cd Autoware/ros and ./run

The run time manager showed up on the screen this time :))).

PR #688 (2017-06-12): fix README for Ubuntu 16.04
fix #682
commit 840937b6 (2017-06-12): fix README for Ubuntu 16.04
fix #682