autowarefoundation/autoware #691: Compile error:" make: INTERNAL: Exiting with 10 jobserver tokens available; should be 8! Invoking "make -j8 -l8" failed "

MrYuJing, 2017-06-13:
Hello!
When I reach these instructions section "How to build" on the web page "https://github.com/CPFL/Autoware" and then other command are done success but the last one "./catkin_make_release " failed. Something wrong like this:
"
[ 65%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_mapping.dir/nodes/common.cpp.o
The bug is not reproducible, so it is likely a hardware or OS problem.
make[2]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/nodes/orb_matching/orb_matching.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
Linking CXX executable /home/yj/Autoware/ros/devel/lib/orb_localizer/orb_mapping_offline
[ 65%] Built target orb_mapping_offline
make[1]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX executable /home/yj/Autoware/ros/devel/lib/orb_localizer/orb_mapping
[ 65%] Built target orb_mapping
make: *** [all] Error 2
make: INTERNAL: Exiting with 10 jobserver tokens available; should be 8!
Invoking "make -j8 -l8" failed
"
Can anyone here help me fix it out? Thanks a lot!

sujiwo, 2017-06-13:
Could you specify OS and ROS version you use?

MrYuJing, 2017-06-13:
OS: 14.04 LTS
ROS: indigo

sujiwo, 2017-06-13:
I'm maintainer for orb_slam in Autoware. Can you try to isolate the problem by compile orb_slam separately using this command

catkin_make --pkg orb_localizer

MrYuJing, 2017-06-13:
Nice! It's worked well. Thanks!
And then what should I do?

sujiwo, 2017-06-13:
Looks like a bug of GNU Make. Lets try running make with fewer process:

./catkin_make_release --make-args -j3

MrYuJing, 2017-06-13:
OK, try it right now

MrYuJing, 2017-06-13:
Wow, everything is OK!
Thanks for your kindly help!

closed 2017-06-13 by sujiwo