autowarefoundation/autoware #693: Localization is not completed.

andoh104, 2017-06-13:
Localization is not completed. This is a Rviz screen shot.
[image]

After this commit, Localization is not completed.
https://github.com/CPFL/Autoware/commit/409e75adcc26e6966c1516d33929d95ba054aa91
Merge pull request #661 from andoh104:fixDocker

Before this commit, Localization is completed.
https://github.com/CPFL/Autoware/commit/e05cf7b7d6c065fa3f4d7abe92ba0f3f9a4f753c
Missing package of Dockerfile #652

andoh104, 2017-06-14:
* At tag 1.2.0, Localization is not completed.
https://github.com/CPFL/Autoware/tree/1.2.0

andoh104, 2017-06-15:
diff e05cf7b and 409e75a:
https://github.com/CPFL/Autoware/compare/e05cf7b7d6c065fa3f4d7abe92ba0f3f9a4f753c...409e75adcc26e6966c1516d33929d95ba054aa91

andoh104, 2017-06-15:
I tested some revision between e05cf7b and 409e75a .

So, I think this commit causes localization problem.
https://github.com/CPFL/Autoware/commit/14f7eca8a5b5aecf110ce6ebb3ab0467323b75fd

andoh104, 2017-06-17:
I tried to debug this issue, but I could only make a temporary patch. This patch will make  the commit 14f7eca unavailable. 

```
~/Autoware$ git diff
diff --git a/ros/src/sensing/filters/packages/points_downsampler/include/points_
index b42336e..2e61291 100644
--- a/ros/src/sensing/filters/packages/points_downsampler/include/points_downsam
+++ b/ros/src/sensing/filters/packages/points_downsampler/include/points_downsam
@@ -21,7 +21,11 @@ static pcl::PointCloud<pcl::PointXYZI> removePointsByRange(pc
     }
   }
 
+#if 0
   return narrowed_scan;
+#else
+  return scan;    //  This is a only tempolary patch for Localization problem.
+#endif
 }
 
 #endif // POINTS_DOWNSAMPLER_H
```

kitsukawa, 2017-06-18:
The commit 14f7eca does not include the modification of matching algorithm. In our environment, we can do matching as before. By the way, do you open the app window every time after running ndt_matching? The problem may be solved by opening app window and clicking OK after running ndt_matching.

andoh104, 2017-06-19:
After checking "ring_filter" of Runtime Manager, Localization is completed.
[image]

But I think this issue should be fixed.

andoh104, 2017-06-26:
I committed this temporary patch into my branch.
https://github.com/andoh104/Autoware/blob/LocalizationPatch/ros/src/sensing/filters/packages/points_downsampler/include/points_downsampler.h

andoh104, 2017-06-30:
@kitsukawa -san

I debuged this method:removePointsByRange.
https://github.com/andoh104/Autoware/blob/LocalizationPatch/ros/src/sensing/filters/packages/points_downsampler/include/points_downsampler.h
Then I found that max_range is always set to 0.0 . So this is bug.

andoh104, 2017-07-02:
I have made a Pull Request to fix this issue.

kitsukawa, 2017-07-13:
@andoh104 Thank you very much for the bug fixes.

andoh104, 2017-07-13:
Don't mention it.
I will close this, if it is merged into master branch.
Thanks,

andoh104, 2017-07-15:
It has been merged into master branch.
I tested master branch is OK.

closed 2017-07-15 by andoh104