autowarefoundation/autoware #694: Calibration Tool Kit flash back wenn grab

Stella-Chen, 2017-06-13:
Hello. I'm trying to use Calibration Tool Kit with an USB camera (Microsoft LifeCam Studio), a checkerboard (Pattern Size 2.5cmx2.5cm, Pattern Number 8x6) to do camera intrinsic calibration. The input image topic is "image_raw", the calibration type is "camera only".
The image looks well, but if I push the Button "Grab", the Calibration Tool Kit would flash back. Only "_error: (-210) The source image must be 8-bit single-channel (CV_8UC1) in function cvFindCornerSubPix_" is displayed in runtime_manager window. What confusing is, if the whole checkerboard was not seen in the image, the window would not be closed after I pushed "Grab".
This problem exists at all times, no matter I use real time image or rosbag.
All Tutorials of Calibration Tool Kit I found online are about calibration between camera and Lidar. Could anyone tell me, have I done something wrong and how to do "camera only" calibration in Autoware?

annaborn, 2017-06-15:
Had the same problem. I fixed by adding the image colour conversion in the file Autoware/ros/src/sensing/fusion/packages/calibration_camera_lidar/CalibrationToolkit/calibrationtoolkit.cpp. In the file calibrationtoolkit.cpp in function `bool CalibrateCameraChessboardROS::refreshImage()` (line 491) add:
```
    cv_bridge::CvImagePtr cv_image = cv_bridge::toCvCopy(msg, sensor_msgs::image_encodings::BGR8);
    cv::Mat tempimage = cv_image->image.clone();
    cv::cvtColor(tempimage, calibimage, CV_BGR2GRAY);
```

Stella-Chen, 2017-06-20:
Sorry. This suggestion doesn't work for me. The image in Calibration Tool Kit is still colorful. No matter I use Rosbag or online image from the camera, the flash-back problem is still there. Now I make "calibration.ymal" file directly with ros for other works. But I am still struggling with this flash-back problem...

annaborn, 2017-06-20:
Ok. I see there are much more functions to refresh an image depending on what you want to calibrate: camera only, camera+lidar, camera+velodyne. I would suggest, that you look for all functions that write some image into `calibimage` variable. Because I don't know which `refreshImage` function you are using at the moment. 
For the camera calibration the function `bool CalibrateCameraChessboardROS::refreshImage()` will be called, but if you are trying to calibrate the camera and lidar/velodyne than it will probably call `bool CalibrateCameraVelodyneChessboardROS::refreshImage()` or `bool CalibrateCameraLidarChessboardROS::refreshImage()`. So the conversion to grayscale should be added to all of the function.
But if you added the conversion and the calibration viewer still shows you the rgb image, it seems that not the last compilied version of calibration package is used.

Stella-Chen, 2017-06-20:
Thank you! After building the package, it works perfect!

closed 2017-06-24 by Stella-Chen