autowarefoundation/autoware #695: what's the physical meaning of “kappa” in the package waypoint follower

ipatient-zju, 2017-06-15:
when I learn the velocity control in Autoware, I found the param calculated in function calcCurvature in pure_pursuit.cpp of package  waypoint follower as follows,
`
    double denominator = pow(getPlaneDistance(target, current_pose_.position), 2);
    double numerator = 2 * calcRelativeCoordinate(target, current_pose_).y;
    kappa = numerator / denominator;
`
How to explain the param "kappa" ?

travelbureau, 2017-06-15:
Curvature: rad/m. Tells you rate of change of rotation with respect to arc
length of path. Directly proportional to twist command.

ahuer2435, 2018-04-23:
计算曲率：kappa
这个计算公式貌似和标准计算公式不同。
曲率计算公式:R=L^2/(2*x),R 为曲率半径,其倒数为对应的曲率.参考论文:一种纯追踪模型改进算法和https://zh.wikipedia.org/zh-hans/%E6%9B%B2%E7%8E%87
L是车辆形式的圆弧的弦长,x是目标点在车辆坐标系下的横坐标.车辆坐标是以前向为y轴,右侧为x轴.圆弧的两点是车辆的当前位置和目标点的位置.
getPlaneDistance(target, current_pose_.pose.position)表示弦长L.
calcRelativeCoordinate(target, current_pose_.pose).y: 目标点target在车体坐标系下的横坐标,感觉应该是x,不是y,这个应该是和车辆坐标系的规定有关，如果是y，车前向应该是x轴；如果是x，车前向应该是y轴，一般似乎都是后一种形式。

kfunaoka, 2018-07-13:
Answered by travelbureau and ahuer2435

closed 2018-07-13 by kfunaoka