autowarefoundation/autoware #698: Installing dependencies fails

keenan-burnett, 2017-06-16:
oracle-java7-installer no longer works, this is causing many of the dependency installs to fail.

dejanpan, 2017-07-10:
Can you update this with:

which install instructions did you follow?
which ubuntu version do you use (14.04, 15.04, 16.04)?
which autoware version (e.g. master/develop branch)?
which ROS version?

I've got ubuntu 14.04, master branch and ROS indigo and don't even have oracle-java7-installer deb.

keenan-burnett, 2017-07-17:
I followed the install instructions from your manual.
14.04
master
Jade

dejanpan, 2017-07-17:
thx @keenan-burnett. we need to start releasing autoware on http://wiki.ros.org/build.ros.org to avoid regressions like this. I will talk to @yk-fujii about this in the next days.

kfunaoka, 2018-07-13:
Obsolete? Please try one of the following procedure.

https://github.com/CPFL/Autoware/wiki/Docker
https://github.com/CPFL/Autoware/wiki/Source-Build

closed 2018-07-13 by kfunaoka