autowarefoundation/autoware #702: Change transform matrix

Stella-Chen, 2017-06-21:
Hello. Has anyone tried to change the transform matrix between Velodyne and GNSS manually? We have tried to make a map with ndt_mapping. But there is an obvious difference between the Velodyne data and GNSS data, exactly, it's a rotation about 90 degrees (this difference is shown in the mapping result, not the coordinates being shown in Rviz). Now we are using the demo transform data with "Autoware/ros/src/util/packages/runtime_manager/scripts/setup_tf.launch". And we think, maybe this problem appears because the transform matrix from our GPS to base_link is not the same as in the demo. So we want to know, is it possible to change the transform matrix manually?

amc-nu, 2017-07-20:
@Stella-Chen 
Please have a look at
[http://wiki.ros.org/tf#static_transform_publisher](http://wiki.ros.org/tf#static_transform_publisher)

closed 2017-07-20 by amc-nu

alva786, 2019-05-16:
@Stella-Chen
I meet the same problem with you.
How do you solve it?
Could you please share your experience?  Thank you