autowarefoundation/autoware #704: Use GPS data in ndt mapping

miketry0313, 2017-06-23:
Hi,

Thanks for open source your code to the community.

I am trying to implement ndt mapping on my own sensor measurement. 

One thing I have observed that it is only using imu and wheel odometry to build the map in ndt_mapping.cpp but no GPS data.

If my above observation is correct, is it helpful to integrate GPS data in the script to avoid the long term drift from imu and odom measurement?

Thanks.

Mike

thomas3016, 2017-08-21:
Anybody have get idea?

asimay, 2017-11-24:
it's already support GPS data.

dejanpan, 2017-11-25:
@andoh104 @shinpei0208 say gps is currently only used for the initialization. it would be fairly easy to add it into the actual mapping process too. Contributions are welcome and let us know if you need some guidance @miketry0313

closed 2020-05-29 by gbiggs