autowarefoundation/autoware #705: Unable to launch kittiplayer.launch

amsks, 2017-06-23:
I have followed all the steps given in readme to the point. However, when I try to run the kittiplayer.launch using the roslaunch command, as error is displayed saying 'can't find node kitti_player in the package kitti_player'.

ebeowulf, 2017-06-28:
Which ROS version are you using? kitti_player is not built by default in ROS kinetic, and getting it to run is more than a little painful

sarahisyoung, 2017-07-25:
How would I run it on kinetic?

Fancy93, 2017-09-26:
@ebeowulf Does  kitti_player  can  be used in indigo?

amc-nu, 2017-09-26:
When using the launch file make sure you’re in a sourced terminal before executing the command.

After compiling Autoware successfully, while on the `Autoware/ros` directory execute:
`$source devel/setup.bash` 
Then execute the roslaunch command as specified in the README.

aztinki, 2017-09-27:
@amc-nu I am trying to use the kitti datasets and I am not sure how to proceed after launching the kitti player as described in the readme file. Could you describe the next steps?
Also, I go this when I launched the kitti player
[ERROR] [1506440991.917541813]: Error reading CAMERA02/CAMERA03 calibration
[ERROR] [1506440991.926335961]: Error reading CAMERA00/CAMERA01 calibration
[kitti_box_publisher-7] process has died [pid 8959, exit code 1, cmd /scratch/user/Autoware/ros/src/util/packages/kitti_pkg/kitti_box_publisher/scripts/kitti_box_publisher.py /scratch/user/Autoware/ros/src/util/packages/kitti_pkg/kitti_player/dataset/2011_09_26/2011_09_26_drive_0001_sync//tracklet_labels.xml __name:=kitti_box_publisher __log:=/scratch/user/.ros/log/550793de-a2d2-11e7-ac7a-9c5c8ec0922f/kitti_box_publisher-7.log].
log file: /scratch/user/.ros/log/550793de-a2d2-11e7-ac7a-9c5c8ec0922f/kitti_box_publisher-7*.log
[kitti_player-8] process has finished cleanly

amc-nu, 2017-09-27:
@aztinki Looks like it's working correctly. Only Camera00 calibration is being used by Autoware, you can safely ignore the calibration related errors.
Did you try to display in RViz or echo the published data (Step 5).
https://github.com/CPFL/Autoware/tree/master/ros/src/util/packages/kitti_pkg

If you still cannot see anything, the data provided is not enough to help you solve the problem. Enable output to screen inside your launch file (http://wiki.ros.org/roslaunch/XML/node), since the errors are not being shown, but logged to the file indicated in your post.
[image]

aztinki, 2017-09-27:
I did echo it and got various data.. But I did not visualize using Rviz. I wanted to run the Autoware software, and I don't know how to use the kitti player to get the simulation you have shown in the image. How do i get the published topics from the kitti player to feed the runtime manager?

aztinki, 2017-09-27:
I suppose, I bag the topics and use it to simulate?

amc-nu, 2017-09-27:
If you can echo the data it means it’s working. just add it to Rviz.
For pointcloud follow the instructions from here
https://answers.ros.org/question/49744/how-to-visualize-point-cloud/

For image data
http://wiki.ros.org/rviz/DisplayTypes/Camera

And then you’ll see the data just like the screen I shared above.

aztinki, 2017-10-10:
@amc-nu thank you very much! I am able to visualize as well as run object detection on Kitti datasets.

amc-nu, 2017-10-10:
Great!

closed 2017-10-10 by amc-nu