autowarefoundation/autoware #710: ROSBAG data of Moriyama video?

k22jung, 2017-06-27:
I am trying to view the video of the ROSBAG data of Moriyama driving. Does a video exist in the .bag data? I followed the steps in setting up object detection in the tutorials but there's no option to select anything in RViz. What are the correct steps in doing this?

Stella-Chen, 2017-06-27:
I think, there are camera data in the Moriyama's ROSBAG integrated. If you look at the information of Moriyama's ROSBAG while playing, you would see a topic called /image_raw. In RViz you could add ImageViewerPlugin by clicking "Panels -> Add New Panel -> integrated_viewer -> ImageViewerPlugin" or just add an image topic. And you'll see the video in your RViz. Hope it will be helpful. :)

k22jung, 2017-06-28:
Here is a screenshot on RViz. Under "topics", there's no "image_raw". Are you sure it's there?
[image]

amc-nu, 2017-06-28:
There is no image data in any of the demo rosbags due to privacy concerns.

closed 2017-06-28 by k22jung