autowarefoundation/autoware #715: Autoware Installation

Ayushas2427, 2017-06-28:
On installing the AutoWare package, i am getting the following error:

[image]

I have installed the pcl version 1.8.0.
Thanks

dejanpan, 2017-07-10:
Can you update this with:
- which install instructions did you follow?
- which ubuntu version do you use (14.04, 15.04, 16.04)?
- which autoware version (e.g. master/develop branch)?

Ayushas2427, 2017-07-10:
Respected,
Thank you fir the reply.
1: I used the following set of commands s given in the github link:
```
% sudo apt-get install ros-kinetic-desktop-full ros-kinetic-nmea-msgs ros-kinetic-nmea-navsat-driver ros-kinetic-sound-play ros-kinetic-jsk-visualization ros-kinetic-grid-map ros-kinetic-gps-common
% sudo apt-get install ros-kinetic-controller-manager ros-kinetic-ros-control ros-kinetic-ros-controllers ros-kinetic-gazebo-ros-control ros-kinetic-joystick-drivers
% sudo apt-get install libnlopt-dev freeglut3-dev qtbase5-dev libqt5opengl5-dev libssh2-1-dev libarmadillo-dev libpcap-dev gksu libgl1-mesa-dev libglew-dev python-wxgtk3.0

$ cd $HOME
$ git clone https://github.com/CPFL/Autoware.git
$ cd ~/Autoware/ros/src
$ catkin_init_workspace
$ cd ../
$ ./catkin_make_release
```

2: I am using ubuntu 14.04
3: I simply followed the link of [github](https://github.com/CPFL/Autoware) and so I am not sure of which autoware version I am using.
4: ROS Vesrion: Indigo

Thank you
Regards

dejanpan, 2017-07-10:
If you ran `sudo apt-get install ros-kinetic-desktop-full ` and the rest then you are NOT using Indigo but Kinetic.
For Autoware version cd into where you cloned Autoware and run `git log` and copy the top most git commit sha.

Ayushas2427, 2017-07-20:
Respected Sir,
I regret, I mentioned the wrong commands in the comment above.
I have corrected the commands now.

As said before, I am using ROS Indigo only.
Thanks

amc-nu, 2018-04-07:
Autoware doesn't support PCL 1.8 yet.

closed 2018-04-07 by amc-nu