autowarefoundation/autoware #716: Invoking "make -j8" failed

SDwaiwai, 2017-06-28:
My system is Ubuntu16.04
ROS kinetic
When catkin_make_release,it failed.
[ 63%] Building CXX object computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/nodes/common.cpp.o
In file included from /home/sdwaiwai/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/nodes/orb_matching/orb_matching.cpp:10:0:
/home/sdwaiwai/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/nodes/orb_matching/Matcher.h:31:33: fatal error: orb_localizer/debug.h: No such file or directory
compilation terminated.
computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/build.make:62: recipe for target 'computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/nodes/orb_matching/orb_matching.cpp.o' failed
make[2]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/nodes/orb_matching/orb_matching.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/sdwaiwai/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/nodes/orb_matching/Matcher.cpp:8:0:
/home/sdwaiwai/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/nodes/orb_matching/Matcher.h:31:33: fatal error: orb_localizer/debug.h: No such file or directory
compilation terminated.
computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/build.make:86: recipe for target 'computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/nodes/orb_matching/Matcher.cpp.o' failed
make[2]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/nodes/orb_matching/Matcher.cpp.o] Error 1
CMakeFiles/Makefile2:17277: recipe for target 'computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/all' failed
make[1]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j3" failed

sujiwo, 2017-06-29:
Sorry @SDwaiwai, I couldn't reproduce the error in my system (16.04). Are you sure you are using latest version of develop branch?
Can you please try this command:
`catkin_make --pkg orb_localizer`

anndymaktub, 2017-07-07:
To make sure debug.h has been generated before building orb_matching/orb_matching_pf.
You need to add the following codes  into `orb_localizer/CMakeLists.txt`

```
add_executable (
        orb_matching
                nodes/orb_matching/orb_matching.cpp
                nodes/orb_matching/Matcher.cpp
#               nodes/orb_matching/FusionOdometry.cpp
                nodes/common.cpp
)

ADD THIS: dependencies(orb_matching orb_localizer_generate_messages_cpp)
```

```
add_executable (
        orb_matching_pf
                nodes/orb_matching/orb_matching_pf.cpp
                nodes/orb_matching/Matcher.cpp
                nodes/orb_matching/FusionOdometry.cpp
                nodes/common.cpp
)
ADD THIS:  add_dependencies(orb_matching_pf orb_localizer_generate_messages_cpp)
```

GuillaumeHauss, 2017-10-26:
hello, 
I'm running Ubuntu 16.04 and ROS Kinetic, and I ran into the same error as described by @SDwaiwai . However, after checking the CMakeLists.txt file, I do have the fix 
`add_executable (
	orb_mapping
		nodes/orb_mapping/orb_mapping.cpp
		nodes/common.cpp
)

target_link_libraries (
	orb_mapping
		${ORB_BIN_LINKS}
		${LINK_LIBRARIES}	
)

add_executable (
	orb_matching
		nodes/orb_matching/orb_matching.cpp
		nodes/orb_matching/Matcher.cpp
#		nodes/orb_matching/FusionOdometry.cpp
		nodes/common.cpp
)

add_dependencies(orb_matching orb_localizer_generate_messages_cpp)

target_link_libraries (
	orb_matching
		${ORB_BIN_LINKS}
		${LINK_LIBRARIES}
)

add_executable (
	orb_matching_pf
		nodes/orb_matching/orb_matching_pf.cpp
		nodes/orb_matching/Matcher.cpp
		nodes/orb_matching/FusionOdometry.cpp
		nodes/common.cpp
)

add_dependencies(orb_matching_pf orb_localizer_generate_messages_cpp)

target_link_libraries (
	orb_matching_pf
		${ORB_BIN_LINKS}
		${LINK_LIBRARIES}
)`

The output of catkin_make_release clearly mentions this package. 
`CMakeFiles/Makefile2:16991: recipe for target 'computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_mapping_offline.dir/all' failed
make[1]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_mapping_offline.dir/all] Error 2
`
and
`CMakeFiles/Makefile2:17447: recipe for target 'computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/all' failed
make[1]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching.dir/all] Error 2
`
and 
`CMakeFiles/Makefile2:17669: recipe for target 'computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching_pf.dir/all' failed
make[1]: *** [computing/perception/localization/packages/orb_localizer/CMakeFiles/orb_matching_pf.dir/all] Error 2
`

Is there another fix available ?

dejanpan, 2017-11-09:
@GuillaumeHauss can you please say which exact branch of Autoware do you use, which Ubuntu and how do you compile it? 

Also can you provide full compile backtrace?

GuillaumeHauss, 2017-11-09:
Hi, 
I'm using the master branch, as it was the default one when I downloaded the source code.
As for the Ubuntu Version, I'm running 16.04 Xenial
For compilation, I'm using the `catkin_make_release` script provided with the source code. 

I think it'll be best to gather all discussion on that matter here and discard this [issue](https://github.com/CPFL/Autoware/issues/862) then. As you wish. 

I'm out of office until next Monday, I'll send you the full compile backtrace then. 

Cheers

closed 2018-04-07 by amc-nu