autowarefoundation/autoware #719: Any tips on using calibration toolkit?

bobjohnson2040, 2017-07-01:
Do you have any suggestions on how to use calibration toolkit? I am trying to calibrate my Velodyne VLP-16 and Pointgrey camera. How many images/scans do you recommend grabbing/annotating?

I'm having 2 issues:

1. The main issue is the calibration toolkit frequently crashes and I have to start over. This happens the most when trying to save the calibration results.

2. I'm having a difficult time accurately projecting the red points onto the image. 

I really would appreciate some help on this. I have been trying to get the calibration toolkit to work for over a week. Thank you!!!

dejanpan, 2017-07-10:
I unfortunately do not have an answer beyond this video: https://www.youtube.com/watch?v=pfBmfgHf6zg.

Out of curiosity, which camera and which lidar sensor do you use?

bobjohnson2040, 2017-07-14:
@dejanpan I would greatly appreciate any help possible.

I have watched that video numerous times.

I am using a Velodyne VLP16 and a Pointgrey Flea3 camera with a wide angle lens.

Do you know if I'm supposed to calibrate the camera by itself before doing the `Camera -> Velodyne` calibration? Should I use the Autoware camera calibration or the ROS camera_calibration package?

I noticed from the discussion between @Eric-Gonzalez and @alexanderhmw on issue #498 that Eric published `/camera/camera_info` when doing `Camera -> Velodyne`. Is this required(the readme doesn't mention it)?

Also how many Velodyne/Camera 'Grab' readings should I do?

THANK YOU!!!!

dejanpan, 2017-07-19:
@bobjohnson2040 I will give it a try tomorrow, till tomorrow I am not behind a Linux computer.

I found this document it does not talk about /camera/camera_info: https://github.com/CPFL/Autoware/blob/6aea989f336561b94e5c29becccbd47228cd17f3/ros/src/sensing/fusion/packages/calibration_camera_lidar/CalibrationToolkit_Manual.pdf.

However I am pretty sure that they use intrinsically calibrated camera (which is the information that /camera/camera_info carries.

amc-nu, 2017-07-19:
@bobjohnson2040 The calibration Toolkit will find both he intrinsic and extrinsic calibration matrices
so there is no need to calibrate it beforehand.

We usually grab 70-100 frames

bobjohnson2040, 2017-07-25:
@dejanpan @amc-nu Thanks guys!! 

Do you know if a wide angle lens could be the cause of this? Are either of you using a wide angle lens?

cwee9616, 2017-10-31:
@dejanpan 

Hi I am trying to calibrate 100 degree FOV camera with vlp16 using autoware. the results are not as good as for the smaller angle cameras. please share with me if you have any information in this regard. I am not sure if Autoware is suitable for wide angle extrinsic calibrations.

dejanpan, 2017-10-31:
@cwee9616 we will get our car in 3 weeks and then do the calibration as a first thing and I can only then tell you more. Our wide angle camera is 70deg though.

@amc-nu  any thoughts?

cwee9616, 2017-10-31:
@dejanpan thanks for the reply. for 70 degree cams Autoware would work fine. I have tested it with a 65 degree one.
with the 100 degree one extrinsic calibratiion is leass accurate. hence i tried feeding the undistorted image for calibration this improved the accuracy but still has some miss alignment with the velodyne.other issue wit this is Autoware recalibrate the intrinsics for the undistorted image internally.

cwee9616, 2017-11-05:
@bobjohnson2040 
Hi did you able to calibrate the wide angle camera with the vlp16? which tool did u used?I have tried Autoware but it seems that it is not good enough for wide angle calibration.

amc-nu, 2018-04-07:
Please try new calibration tool. #1131

closed 2018-04-07 by amc-nu

GeeKmyself, 2018-08-08:
did u fixed it , i have the some problem like u do