autowarefoundation/autoware #722: VectorMap Server Call failed

andoh104, 2017-07-09:
This error message was output .
> vectormap_filtering: VectorMap Server Call failed. Make sure vectormap_server is running. No filtering performed.

The cause of this error message is as follows:
ros/src/data/packages/vector_map_server/nodes/vector_map_server/vector_map_server.cpp
ros::ServiceServer is_way_area_srv = nh.advertiseService("vector_map_server/is_way_area",
&VectorMapServer::isWayArea, &vms);

The service of "vector_map_server/is_way_area" is not advertised by vector_map_server.

andoh104, 2017-07-09:
The service  "vector_map_server/is_way_area"  don't exist. I think the correct service name is "vector_map_server/PositionSate".

But those services are not running.
```
@:~/Autoware$ rosservice list
/base_link_to_localizer/get_loggers
/base_link_to_localizer/set_logger_level
/calibration_publisher/get_loggers
/calibration_publisher/set_logger_level
/driver_nodelet/get_loggers
/driver_nodelet/set_logger_level
/euclidean_cluster/get_loggers
/euclidean_cluster/set_logger_level
/euclidean_lidar_track/get_loggers
/euclidean_lidar_track/set_logger_level
/joint_state_publisher/get_loggers
/joint_state_publisher/set_logger_level
/map_to_mobility/get_loggers
/map_to_mobility/set_logger_level
/ndt_matching/get_loggers
/ndt_matching/set_logger_level
/nmea2tfpose/get_loggers
/nmea2tfpose/set_logger_level
/obj_car/dpm/get_loggers
/obj_car/dpm/set_logger_level
/obj_car/fusion/get_loggers
/obj_car/fusion/set_logger_level
/obj_car/ranging/get_loggers
/obj_car/ranging/set_logger_level
/obj_car/reprojection/get_loggers
/obj_car/reprojection/set_logger_level
/obj_car/tracking/get_loggers
/obj_car/tracking/set_logger_level
/play_1499583673410647840/get_loggers
/play_1499583673410647840/set_logger_level
/points2image/get_loggers
/points2image/set_logger_level
/points_map_loader/get_loggers
/points_map_loader/set_logger_level
/relay_camera_info/get_loggers
/relay_camera_info/set_logger_level
/robot_state_publisher/get_loggers
/robot_state_publisher/set_logger_level
/rosout/get_loggers
/rosout/set_logger_level
/rqt_gui_py_node_6658/get_loggers
/rqt_gui_py_node_6658/set_logger_level
/runime_manager_2697_1499583660203/get_loggers
/runime_manager_2697_1499583660203/set_logger_level
/vector_map_loader/get_loggers
/vector_map_loader/set_logger_level
/velodyne_nodelet/get_loggers
/velodyne_nodelet/set_logger_level
/velodyne_nodelet/set_parameters
/velodyne_nodelet_manager/get_loggers
/velodyne_nodelet_manager/list
/velodyne_nodelet_manager/load_nodelet
/velodyne_nodelet_manager/set_logger_level
/velodyne_nodelet_manager/unload_nodelet
/voxel_grid_filter/get_loggers
/voxel_grid_filter/set_logger_level
/vscan_filling/get_loggers
/vscan_filling/set_logger_level
/world_to_map/get_loggers
/world_to_map/set_logger_level
@:~/Autoware$
```

kuriking, 2017-07-09:
You must run vector_map_server.

```
$ rosrun vector_map_server vector_map_server

$ rosservice list | grep /vector_map_server
/vector_map_server/get_cross_road
/vector_map_server/get_cross_walk
/vector_map_server/get_curb
...
/vector_map_server/is_way_area
/vector_map_server/set_logger_level
```

andoh104, 2017-07-09:
Thank you very much. I have been able to run vector_map_server as follows.
~~~
Autoware/ros$ source devel/setup.bash
Autoware/ros$ rosrun vector_map_server vector_map_server &
[1] 19164
Autoware/ros$ rosservice list | grep /vector_map_server
/vector_map_server/PositionState
/vector_map_server/get_cross_road
/vector_map_server/get_cross_walk
/vector_map_server/get_curb
/vector_map_server/get_curve_mirror
/vector_map_server/get_drive_on_portion
/vector_map_server/get_dtlane
/vector_map_server/get_fence
/vector_map_server/get_guard_rail
/vector_map_server/get_gutter
/vector_map_server/get_lane
/vector_map_server/get_loggers
/vector_map_server/get_node
/vector_map_server/get_rail_crossing
/vector_map_server/get_road_edge
/vector_map_server/get_road_mark
/vector_map_server/get_road_pole
/vector_map_server/get_road_sign
/vector_map_server/get_side_strip
/vector_map_server/get_side_walk
/vector_map_server/get_signal
/vector_map_server/get_stop_line
/vector_map_server/get_street_light
/vector_map_server/get_utility_pole
/vector_map_server/get_wall
/vector_map_server/get_way_area
/vector_map_server/get_white_line
/vector_map_server/get_zebra_zone
/vector_map_server/set_logger_level 
~~~

And I understood that service type is "vector_map_server/PositionState", and service name is "vector_map_server/is_way_area".
But I think RuntimeManager should run  vector_map_server.

andoh104, 2017-07-25:
I have made a pull request (https://github.com/CPFL/Autoware/pull/739) .

andoh104, 2017-09-21:
It has been merged into the master branch.

closed 2017-09-21 by andoh104