autowarefoundation/autoware #724: Issues with Autoware master branch.

dejanpan, 2017-07-10:
Today I installed and tried out master brnach of Autoware (70bd1c2d7adbb6c8f8c859d75a46c3af3a853f3a). 
I am running on Ubuntu 14.04, ROS Indigo, using ample_moriyama_150324.bag.

I encountered the following issues:
Compilation: https://github.com/CPFL/Autoware/pull/723
Running:
- ndt_matching is off in yaw direction:
[image]

- car is tilted: 
[image]

- running Gazebo fails with the following message:
```
Gazebo multi-robot simulator, version 2.2.3
Copyright (C) 2012-2014 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Msg Waiting for masterGazebo multi-robot simulator, version 2.2.3
Copyright (C) 2012-2014 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Msg Waiting for masterMsg Connected to gazebo master @ http://127.0.0.1:11345
Msg Publicized address: 10.0.0.72
..[ INFO] [1499653025.473772218]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1499653025.474143987]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...

Warning [ModelDatabase.cc:334] Getting models from[http://gazebosim.org/models/]. This may take a few seconds.
Msg Connected to gazebo master @ http://127.0.0.1:11345
Msg Publicized address: 10.0.0.72

Warning [gazebo.cc:215] Waited 1seconds for namespaces.
Warning [gazebo.cc:215] Waited 1seconds for namespaces.
Warning [gazebo.cc:215] Waited 1seconds for namespaces.
Warning [gazebo.cc:215] Waited 1seconds for namespaces.
Warning [gazebo.cc:215] Waited 1seconds for namespaces.
Warning [gazebo.cc:215] Waited 1seconds for namespaces.
Warning [gazebo.cc:215] Waited 1seconds for namespaces.
Warning [gazebo.cc:215] Waited 1seconds for namespaces.
Warning [gazebo.cc:215] Waited 1seconds for namespaces.
Warning [gazebo.cc:215] Waited 1seconds for namespaces.
Error [gazebo.cc:220] Waited 11 seconds for namespaces. Giving up.
Error [Node.cc:90] No namespace found
Error [Node.cc:90] No namespace found
Error [Node.cc:90] No namespace found
Error [Node.cc:90] No namespace found
Error [Node.cc:90] No namespace found
Error [Node.cc:90] No namespace found
Error [Node.cc:90] No namespace found
Error [Node.cc:90] No namespace found
Error [Node.cc:90] No namespace found
Error [Node.cc:90] No namespace found
Error [Node.cc:90] No namespace found
```

- ndt_matching crashes if you compile in non-Release mode

andoh104, 2017-07-10:
NDT matching problem has been fixed . But it has not been merged into master branch yet.
https://github.com/CPFL/Autoware/issues/693

dejanpan, 2017-07-10:
Ok, thx. I will try out develop then.

closed 2018-01-15 by yk-fujii