autowarefoundation/autoware #736: Runtime manager not starting

iulialexandra, 2017-07-21:
I am trying to run Autoware but only the roscore terminal shows up when I type ./run. I get prompted for my password for the Runtime Manager but after I type it nothing happens, I don't even get an error, so I have no clue how to debug this.

Does anybody know what might be going on?

I am using Ubuntu 16.04.

dejanpan, 2017-07-21:
Which ROS Version are you using and which branch did you check out and
compile (best give us commit sha).

What you need to type in is your root user password.

iulialexandra, 2017-07-21:
I installed ros-kinetic through apt-get.
I followed the README instructions for Autoware and did _git clone https://github.com/CPFL/Autoware.git_, so I guess it's the master branch. 

I did type the root user password.

dejanpan, 2017-07-21:
I am on master (bf694eb1dfd7656ca0889182d0ae783a1ceecdf7), also 16.04 and kinetic and for me runtime_manager GUI shows up without a problem.

Which SHELL and which terminal app are you using?

What happens if you do following in 2 different terminals
```
$ cd $HOME/Autoware/ros/; source devel/setup.bash; roscore
$ cd $HOME/Autoware/ros/; source devel/setup.bash; rosrun runtime_manager runtime_manager_dialog.py
```

amc-nu, 2017-07-22:
@iulialexandra
It might be a silly question but, did you compile successfully using the catkin_make_release command from a terminal?
I've seen this behavior if the files required by runtime manager haven't been compiled.

manato, 2017-07-23:
@iulialexandra 
This problem sometimes happens by lack of some python dependencies.
Would you refer #682 ?
Conversations here may help you to find out what the exactly problem is.

iulialexandra, 2017-07-23:
Thanks to all for your answers, it works now. I had the same issue as #682, but I hadn't noticed that thread, sorry.

closed 2017-07-23 by iulialexandra

dejanpan, 2017-07-24:
@iulialexandra @manato 
you just installed `python-wxtools` and runtime manager showed up?

This is weird because if I understand correctly then this line failed: https://github.com/CPFL/Autoware/blob/master/ros/src/util/packages/runtime_manager/scripts/runtime_manager_dialog.py#L33.

This module is installed via `libwxgtk3.0-0v5` debian package and if you do a reverse search:
`aptitude why libwxgtk3.0-0v5` you see that `libwxgtk3.0-0v5` is installed via `python-wxgtk3.0` package which is listed in the README.

I do not have `python-wxtools` installed and runtime manager still works for me.

iulialexandra, 2017-07-24:
@dejanpan 

Yep, installed python-wxtools and runtime manager showed up. 
This is the only thing I did to make it work, not sure what the actual problem was.

pkhunphakdee, 2019-05-23:
@dejanpan I only can run the Autoware runtime manager with the command 
$ cd $HOME/Autoware/ros/; source devel/setup.bash; roscore
$ cd $HOME/Autoware/ros/; source devel/setup.bash; rosrun runtime_manager runtime_manager_dialog.py
I cannot run with the command ./run even if I've already installed python-wxtools
Do you know what is the reason? Also, in the runtime manager, there are many bars that show status of CPUs, these bar have no response in my runtime.

gbiggs, 2019-05-24:
Thanks for using Autoware and for your question. However the issue tracker is for posting confirmed bugs and feature requests.

We ask that you please ask questions at the [ROS Answers website](https://answers.ros.org/questions/ask/?tags=autoware) following our support guidelines: https://github.com/autowarefoundation/autoware/wiki/Support-guidelines#ros-answers. Please pay particular attention to the information we ask you to provide.

datnguyen82, 2019-05-30:
@pkhunphakdee ,

The run.sh file in Autoware/docker/generic seems to be not working. 
I changed the script to: 

`#!/bin/sh

XSOCK=/tmp/.X11-unix

XAUTH=/home/$USER/.Xauthority

SHARED_DIR=/home/autoware/.autoware

HOST_DIR=/home/$USER/.autoware

nvidia-docker run \

    -it --rm \
    --volume=$XSOCK:$XSOCK:rw \
    --volume=$XAUTH:$XAUTH:rw \
    --volume=$HOST_DIR:$SHARED_DIR:rw \
    --env="XAUTHORITY=${XAUTH}" \
    --env="DISPLAY=${DISPLAY}" \
    -u autoware \
    --privileged -v /dev/bus/usb:/dev/bus/usb \
    --net=host \
    autoware/autoware:latest-kinetic` 

Then I ran sudo bash run.sh again in Autoware/docker/generic directory and it led me to the docker terminal.

After that, I ran cd Autoware/ros and ./run

The run time manager showed up on the screen this time :))).

datnguyen82, 2019-05-30:
@pkhunphakdee ,

The run.sh file in Autoware/docker/generic seems to be not working. 
I changed the script to: 

`#!/bin/sh

XSOCK=/tmp/.X11-unix

XAUTH=/home/$USER/.Xauthority

SHARED_DIR=/home/autoware/.autoware

HOST_DIR=/home/$USER/.autoware

nvidia-docker run \

    -it --rm \
    --volume=$XSOCK:$XSOCK:rw \
    --volume=$XAUTH:$XAUTH:rw \
    --volume=$HOST_DIR:$SHARED_DIR:rw \
    --env="XAUTHORITY=${XAUTH}" \
    --env="DISPLAY=${DISPLAY}" \
    -u autoware \
    --privileged -v /dev/bus/usb:/dev/bus/usb \
    --net=host \
    autoware/autoware:latest-kinetic` 
According to this guide: https://ai4sig.org/2018/07/docker-for-autoware/?fbclid=IwAR2ImpFvYf8D97HIVQ--YhwZpC4bl6VKS5fZ5uGxixVxjRGjxIbf93V-ywU
Then I ran sudo bash run.sh again in Autoware/docker/generic directory and it led me to the docker terminal.

After that, I ran cd Autoware/ros and ./run

The run time manager showed up on the screen this time :))).

TNAV1, 2019-10-25:
Autoware runtime manager flashes on screen for a second showing
cannot create directory home/xxxx/.rvis': file exists
[captured this on video as it flashed on screen]

then the terminal window show this message.
PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    run (runtime_manager/run)

auto-starting new master
process[master]: started with pid [7568]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 14b8df92-f6bb-11e9-a705-e0d55ea7841f
process[rosout-1]: started with pid [7579]
started core service [/rosout]
process[run-2]: started with pid [7582]
sudo: no tty present and no askpass program specified
[run-2] process has finished cleanly
log file: /home/xxxx/.ros/log/14b8df92-f6bb-11e9-a705-e0d55ea7841f/run-2*.log

this is on autoware 1.12
commands used:
cd autoware.ai
source install/setup.bash
roslaunch runtime_manager runtime_manager.launch