autowarefoundation/autoware #737: Invoking make -j16 failed

claretway, 2017-07-21:
I know that this was covered [here](https://github.com/CPFL/Autoware/issues/716) but I ran into this same issue and the suggestions for debugging did not work for me. I'm using Ubuntu 14.04 with ROS Indigo. I was never able to selectively build the orb_localizer package with `catkin_make --pkg orb_localizer` even after adding the lines to the CMakeLists.txt suggested in the link above.

This is the output I get after trying to build with `./catkin_make_release --make-args -j16` 
```
Scanning dependencies of target pos_db
[  0%] Linking CXX shared library /home/driver/ccui/planner/Autoware/ros/devel/lib/libobj_db.so
[  0%] Building CXX object data/packages/pos_db/CMakeFiles/pos_db.dir/lib/pos_db/SendData.cpp.o
[  0%] Built target obj_db
[  0%] Building CXX object data/packages/pos_db/CMakeFiles/pos_db.dir/lib/pos_db/util.cpp.o
make[1]: *** [sensing/drivers/camera/packages/vectacam/CMakeFiles/vectacam_node.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  0%] Built target kvaser_generate_messages_eus
[  1%] Linking CXX executable /home/driver/ccui/planner/Autoware/ros/devel/lib/map_tools/pcd_binarizer
[  1%] Linking CXX shared library /home/driver/ccui/planner/Autoware/ros/devel/lib/libpos_db.so
[  1%] Built target pos_db
[  1%] Built target pcd_binarizer
[  1%] Building CXX object sensing/drivers/lidar/packages/hokuyo/CMakeFiles/hokuyo_3d.dir/nodes/hokuyo_3d/hokuyo_3d.cpp.o
make[1]: *** [util/packages/fake_drivers/CMakeFiles/fake_camera.dir/all] Error 2
[  1%] Linking CXX executable /home/driver/ccui/planner/Autoware/ros/devel/lib/map_tools/pcd_filter
[  1%] Linking CXX shared library /home/driver/ccui/planner/Autoware/ros/devel/lib/libfastvirtualscan.so
[  1%] Built target fastvirtualscan
[  1%] Built target pcd_filter
[  1%] Linking CXX executable /home/driver/ccui/planner/Autoware/ros/devel/lib/laser_scan_converter/laser_scan_converter_node
[  1%] Built target laser_scan_converter_node
[  2%] Linking CXX shared library /home/driver/ccui/planner/Autoware/ros/devel/lib/libfast_pcl_registration.so
[  2%] Built target fast_pcl_registration
[  2%] Linking CXX executable /home/driver/ccui/planner/Autoware/ros/devel/lib/hokuyo/hokuyo_3d
[  2%] Built target hokuyo_3d
[  2%] Linking CXX executable /home/driver/ccui/planner/Autoware/ros/devel/lib/point_cloud_converter/point_cloud_converter_node
[  2%] Built target point_cloud_converter_node
[  2%] Linking CXX shared library /home/driver/ccui/planner/Autoware/ros/devel/lib/libfast_pcl_filters.so
[  2%] Built target fast_pcl_filters
make: *** [all] Error 2
Invoking "make -j16" failed
```

This is the output I get without any flags, with just `./catkin_make_release `. It appears that I can actually get further without specifying the number of cores also.
```
[ 11%] Built target pcd_arealist
In file included from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34:0,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/LoopClosing.h:34,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/LocalMapping.h:26,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/Tracking.h:33,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/FrameDrawer.h:24,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/Viewer.h:25,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/src/Viewer.cc:21:
/home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/g2o/core/base_binary_edge.h:59:82: warning: ‘Eigen::AlignedBit’ is deprecated (declared at /usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162) [-Wdeprecated-declarations]
       typedef Eigen::Map<Matrix<double, Di, Dj>, Matrix<double, Di, Dj>::Flags & AlignedBit ? Aligned : Unaligned > HessianBlockType;
                                                                                  ^
In file included from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34:0,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/LoopClosing.h:34,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/LocalMapping.h:26,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/Tracking.h:33,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/FrameDrawer.h:24,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/Viewer.h:25,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/src/Viewer.cc:21:
/home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/g2o/core/base_binary_edge.h:60:82: warning: ‘Eigen::AlignedBit’ is deprecated (declared at /usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162) [-Wdeprecated-declarations]
       typedef Eigen::Map<Matrix<double, Dj, Di>, Matrix<double, Dj, Di>::Flags & AlignedBit ? Aligned : Unaligned > HessianBlockTransposedType;
                                                                                  ^
In file included from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:35:0,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/LoopClosing.h:34,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/LocalMapping.h:26,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/Tracking.h:33,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/FrameDrawer.h:24,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/include/Viewer.h:25,
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/src/Viewer.cc:21:
/home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/g2o/core/base_vertex.h:62:74: warning: ‘Eigen::AlignedBit’ is deprecated (declared at /usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162) [-Wdeprecated-declarations]
     typedef Eigen::Map<Matrix<double, D, D>, Matrix<double,D,D>::Flags & AlignedBit ? Aligned : Unaligned >  HessianBlockType;
                                                                          ^
In file included from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34:0,
[... 2 lines cut ...]
                 from /home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/src/Optimizer.cc:22:
/home/driver/ccui/planner/Autoware/ros/src/computing/perception/localization/packages/orb_localizer/Thirdparty/g2o/g2o/core/base_binary_edge.h:59:82: warning: ‘Eigen::AlignedBit’ is deprecated (declared at /usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162) [-Wdeprecated-declarations]
       typedef Eigen::Map<Matrix<double, Di, Dj>, Matrix<double, Di, Dj>::Flags & AlignedBit ? Aligned : Unaligned > HessianBlockType;
                                                                                  ^
[... cut ...]
     typedef Eigen::Map<Matrix<double, D, D>, Matrix<double,D,D>::Flags & AlignedBit ? Aligned : Unaligned >  HessianBlockType;
                                                                          ^
[ 12%] Linking CXX static library /home/driver/ccui/planner/Autoware/ros/devel/lib/libg2o.a
[ 12%] Built target g2o
[ 12%] Linking CXX static library /home/driver/ccui/planner/Autoware/ros/devel/lib/liborb_slam2.a
[ 12%] Built target orb_slam2
make: *** [all] Error 2
Invoking "make -j48 -l48" failed
```

dejanpan, 2017-07-21:
is this master or develop branch?

claretway, 2017-07-21:
I cloned from master

claretway, 2017-07-21:
I was on master

travelbureau, 2017-07-21:
Output says -j48 right before it fails not sure what behavior to expect but
I'm doubting you have 48 cores (although it's not impossible) also -l48
seems suspect perhaps some typo in build script?

<https://github.com/CPFL/Autoware/issues/737#issuecomment-317128067>, or mute
the thread

dejanpan, 2017-07-21:
I have built successfully on 16.04 and kinetic with the same command
```
./catkin_make_release --make-args -j16
```

@amc-nu who is most familiar with build issues in Autoware? Do you guys have any nightly or PR builds for different:
- processor architectures
- different ubuntu versions
- different ROs version
?

claretway, 2017-07-21:
I think there are 24 cores and 24 virtual cores

amc-nu, 2017-07-22:
@claretway
We don't find any problems while building with the script maybe it's something related to your environment.
Please try compiling manually
While in the ros directory:
1. catkin_make clean
2. source devel/setup.bash
3. catkin_make -j8

@dejanpan
afaik, it supports ARM architecture, but that's already included in the scripts.

It also supports ROS hydro or Ubuntu 13.04, 13.10, but that's a very old version of Autoware.

dejanpan, 2017-07-22:
thx. What I meant is do you build on clean versions of all this variations before new pull request gets merged into master?

amc-nu, 2017-07-22:
@dejanpan
We check it builds and works on 14.04 and 16.04.
ARM is also verified.

dejanpan, 2017-07-22:
Do you have any documentation on how building and working is tested? Is it manual?

I am asking because my plan was to release Autoware through ROS build farm: http://wiki.ros.org/bloom/Tutorials/FirstTimeRelease which would check these things where at least building would happen automatically.

amc-nu, 2017-07-22:
@dejanpan
We do it manually on each of our different machines.
We add the new dependencies, clean the environment using the catkin_make_release script and test on each of our different environments before merging to master (since all of us work on the develop branch)

travelbureau, 2017-07-22:
Would definitely try -j4 or -j8. On my 32gb ram workstation I have ran into
memory issues when building with a lot of cores, but sounds like there
might be something else going on as well.

claretway, 2017-07-24:
Okay, now I'm thinking it's an issue with opencv, of which I have version 3.3.0 installed. I got this error, I guess after I ran `catkin_make clean` and built with reduced cores:

```make[2]: *** No rule to make target `/usr/lib/x86_64-linux-gnu/libopencv_videostab.so.2.4.8', needed by `/home/driver/ccui/planner/Autoware/ros/devel/lib/vectacam/vectacam_node'.  Stop.```

claretway, 2017-07-24:
Is Autoware not working with OpenCV3 still @syohex? I guess I missed this. I was under the impression that "OpenCV 2.4.10 or higher" under the README requirements meant that OpenCV3 was supported as well.

claretway, 2017-07-25:
Still running into issues, but earlier issue was due to wrong OpenCV version.

closed 2017-07-25 by claretway