autowarefoundation/autoware #738: Invoking "make -j8 -l8" failed in Ubuntu 16.04

yinchenjia, 2017-07-22:
Here is the last few lines before the build crashs

[ 42%] Generating C++ code from autoware_msgs/CloudCluster.msg
[ 42%] Built target orb_slam2
[ 42%] Generating C++ code from autoware_msgs/ValueSet.msg
[ 42%] Generating C++ code from autoware_msgs/ConfigWaypointFollower.msg
[ 42%] Generating C++ code from autoware_msgs/obj_label.msg
[ 42%] Generating C++ code from autoware_msgs/PointsImage.msg
[ 42%] Generating C++ code from autoware_msgs/ScanImage.msg
[ 42%] Generating C++ code from autoware_msgs/ColorSet.msg
[ 42%] Generating C++ code from autoware_msgs/ConfigCarDpm.msg
[ 43%] Generating C++ code from autoware_msgs/ConfigPedestrianFusion.msg
[ 43%] Generating C++ code from autoware_msgs/vscan_tracked.msg
[ 43%] Generating C++ code from autoware_msgs/centroids.msg
[ 43%] Generating C++ code from autoware_msgs/ConfigCarFusion.msg
[ 43%] Generating C++ code from autoware_msgs/Signals.msg
[ 43%] Generating C++ code from autoware_msgs/vscan_tracked_array.msg
[ 43%] Linking CXX executable /home/motivo/Autoware/ros/devel/lib/udon_socket/udon_receiver
[ 43%] Built target autoware_msgs_generate_messages_cpp
[ 43%] Built target udon_receiver
[ 43%] Linking CXX shared library /home/motivo/Autoware/ros/devel/lib/libop_planner.so
[ 43%] Built target op_planner
[ 43%] Linking CXX executable /home/motivo/Autoware/ros/devel/lib/udon_socket/udon_sender
[ 43%] Built target udon_sender
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

I looked through the previous lines, there are some warning, but I don't really see error. This computer was used for something else in the past. There might be some other ROS package conflicting with Autoware, but I am not sure. Thanks.

dejanpan, 2017-07-22:
please provide full trace, error message is not even here. Also please make sure that you have your packages up to date (apt-get update && apt-get dist-upgrade) and that you only have /opt/ros/kinetic/setup.bash sourced before you clone and compile Autoware.

yinchenjia, 2017-07-23:
It is my fault. Some dependencies are not installed. It is working now. Thanks.

closed 2017-07-23 by yinchenjia

dejanpan, 2017-07-24:
Can you maybe just say which were those dependencies? Are they missing in the README file? Just so we understand if the error was because you didn't fully follow the instructions or because there are things missing in the instructions.