autowarefoundation/autoware #744: /usr/bin/ld error when building

claretway, 2017-07-25:
After running `catkin_make -j8` on my machine (Ubuntu 14.04, ROS Indigo), I can't get any further than this. I have OpenNI installed, and `libDummyDevice.so, libOniFile.so`, and `libPS1080.so` are all in `/usr/lib/OpenNI2/Drivers` but for some reason it's not finding them when I build.

```
/usr/bin/ld: cannot find -lDummyDevice
/usr/bin/ld: cannot find -lOniFile
/usr/bin/ld: cannot find -lPS1080
collect2: error: ld returned 1 exit status
Scanning dependencies of target rosinterface
make[2]: *** [/home/driver/temp/trying/Autoware/ros/devel/lib/kitti_player/kitti_player] Error 1
make[1]: *** [util/packages/kitti_pkg/kitti_player/CMakeFiles/kitti_player.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 64%] Built target pos_uploader
[ 64%] Built target pos_downloader
[ 64%] Built target fusion
[ 64%] Built target lane_detector
[ 64%] Building CXX object util/packages/RobotSDK/rosinterface/CMakeFiles/rosinterface.dir/ROSInterface/rosinterface.cpp.o
[ 66%] Built target orb_slam2
[ 66%] Linking CXX shared library /home/driver/temp/trying/Autoware/ros/devel/lib/libglviewer.so
[ 66%] Built target glviewer
[ 66%] Linking CXX shared library /home/driver/temp/trying/Autoware/ros/devel/lib/librosinterface.so
[ 66%] Built target rosinterface
make: *** [all] Error 2
Invoking "make -j8" failed
driver@hrlh03061-ubuntu:~/temp/trying/Autoware/ros$ 
```

dejanpan, 2017-07-25:
this is something I promised to do for every package but didn't do yet. 
Run this command before catkin_make:
https://github.com/CPFL/Autoware/blob/master/ros/catkin_make_release#L15

dejanpan, 2017-07-25:
@claretway did you manage to fix it?

claretway, 2017-07-25:
@dejanpan Yeah, thanks! It was odd because I've been running that exact script on my local copy and it was failing before. I cloned a fresh one however and it finally worked.

DanielTakeshi, 2019-03-27:
@dejanpan  @claretway Looks like the link is dead. Is there a new link?

albuckley, 2019-04-02:
+1 for dead link

gbiggs, 2019-04-03:
The link is dead because Autoware does not use `catkin` to build any more. We use `colcon` instead now.